agarciadom/ipwebcam-gst #17: Disconnecting immediately

Rika-chan, 2015-04-30:
Debian Jessie

When attempting to connect through USB, it immediately disconnects. When attempting to connect through wifi, it also immediately disconnects.

```
rika-chan@Homunyan:~/Bernkastel$ ./prepare-videochat.sh 
/usr/bin/adb

** (zenity:18138): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.

** (zenity:18143): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
14307

** (zenity:18224): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Setting pipeline to PAUSED ...
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: Failed to query attributes of input 0 in device /dev/video0
Additional debug info:
v4l2_calls.c(134): gst_v4l2_fill_lists (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Failed to get 0 in input enumeration for /dev/video0. (25 - Inappropriate ioctl for device)
Setting pipeline to NULL ...
Freeing pipeline ...

** (zenity:18229): WARNING **: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```

bluezio, 2015-04-30:
Thank you for the report. Unfortunately, I don't have a machine with Debian Jessie installed, so it'll be difficult for me to debug this issue. Did you try the various recommendations in the Troubleshooting section of the script?

Which versions of souphttpsrc, v4l2sink and v4l2loopback are you using?

Rika-chan, 2015-04-30:
v4l2loopback-0.8.0

gir1.2-soup-2.4

I'm unsure which package v4l2sink is in

bluezio, 2015-05-02:
Thanks for the information :-). I'll try and test the script on a Jessie live CD.

bluezio, 2015-05-02:
I've just tried the script in a live CD version of Debian Jessie, and it has worked for me both through adb (installed through android-tools-adb) and Wi-Fi. I'm also using v4l2loopback 0.8.0-4, v4l-utils 1.6.0 and gstreamer-tools 0.10.36-1.5. I'm using these GStreamer plugins as well (checked with "dpkg -l | grep gstreamer.*plugins"):

    ii  gstreamer0.10-plugins-bad:amd64       0.10.23-7.4                         amd64        GStreamer plugins from the "bad" set
    ii  gstreamer0.10-plugins-base:amd64      0.10.36-2                           amd64        GStreamer plugins from the "base" set
    ii  gstreamer0.10-plugins-good:amd64      0.10.31-3+nmu4+b1                   amd64        GStreamer plugins from the "good" set
    ii  gstreamer0.10-x:amd64                 0.10.36-2                           amd64        GStreamer plugins for X11 and Pango
    ii  gstreamer1.0-plugins-bad:amd64        1.4.4-2.1+b1                        amd64        GStreamer plugins from the "bad" set
    ii  gstreamer1.0-plugins-base:amd64       1.4.4-2                             amd64        GStreamer plugins from the "base" set
    ii  gstreamer1.0-plugins-good:amd64       1.4.4-2                             amd64        GStreamer plugins from the "good" set
    ii  gstreamer1.0-plugins-ugly:amd64       1.4.4-2+b1                          amd64        GStreamer plugins from the "ugly" set
    ii  gstreamer1.0-vaapi:amd64              0.5.9-2                             amd64        VA-API plugins for GStreamer
    ii  gstreamer1.0-x:amd64                  1.4.4-2                             amd64        GStreamer plugins for X11 and Pango
    ii  libgstreamer-plugins-bad0.10-0:amd64  0.10.23-7.4                         amd64        GStreamer development files for libraries from the "bad" set
    ii  libgstreamer-plugins-bad1.0-0:amd64   1.4.4-2.1+b1                        amd64        GStreamer development files for libraries from the "bad" set
    ii  libgstreamer-plugins-base0.10-0:amd64 0.10.36-2                           amd64        GStreamer libraries from the "base" set
    ii  libgstreamer-plugins-base1.0-0:amd64  1.4.4-2                             amd64        GStreamer libraries from the "base" set

If all the versions are the same on your system. I'd look at why souphttpsrc is failing on your system. I suppose you tried the simplified pipeline that was in the troubleshooting section, right?

    gst-launch souphttpsrc location="http://$IP:$PORT/videofeed" \
             do-timestamp=true is-live=true \
      ! multipartdemux ! jpegdec ! ffmpegcolorspace ! ximagesink

Rika-chan, 2015-05-03:
Plugins are current, although I have some more plugins,

https://gist.github.com/Rika-chan/4483ff99230557dda335

Output of the simplified pipeline...
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
Error resolving '': No address associated with hostname
Additional debug info:
gstsouphttpsrc.c(924): gst_soup_http_src_finished_cb ():
/GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
libsoup status code 2
Execution ended after 34335218 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

So there does appear to be an error there, although I do not know how to
troubleshoot this

bluezio, 2015-05-03:
The "no address associate with hostname" error message bothers me. Did you specify the same http://host:port/... URL as shown on the phone after starting the IP Webcam server?

Rika-chan, 2015-05-06:
Sorry for late reply.

I made a mistake earlier, and mistyped the IP, whoops.

The simplified pipeline works perfectly fine.

bluezio, 2015-05-06:
Hmm... do you have any webcams plugged into the computer already? Please check if you have any /dev/video* devices before you run the script.

Rika-chan, 2015-05-06:
I do not.

bluezio, 2015-05-06:
OK, then it's not the `souphttpsrc` GStreamer plugin, and it's not due to some other video device confusing things. Let's focus on this part of your original error message:

    Failed to get 0 in input enumeration for /dev/video0. (25 - Inappropriate ioctl for device)

Can you install the `mplayer` package and then run these commands?

    sudo modprobe v4l2loopback
    whoami
    groups
    ls -l /dev/video*
    gst-launch videotestsrc ! v4l2sink device=/dev/video0
    mplayer tv://

If all goes well, you should see something like an old-school TV test signal on mplayer. If mplayer does not play back anything, please post here the output of the various commands. I added the `whoami` and `groups` commands so we could check if you have the right permissions on /dev/video0.

Rika-chan, 2015-05-06:
https://gist.github.com/Rika-chan/45a37e42328c1f166054

Rika-chan, 2015-05-06:
@bluezio 

Ran again, 

```rika-chan@Homunyan:~/Bernkastel$ ls -la /dev/video*
crw-rw----+ 1 root video 81, 0 Apr 30 00:48 /dev/video0
```

bluezio, 2015-05-06:
Nope, it's not the permissions either: you're in the video group and /dev/video0 is group writable.

I'm really sorry for all the work, but I need to find the difference between your system and my Jessie live CD that is causing your issues. Maybe it's the kernel version, or maybe your `/dev/video0` doesn't allow the same video formats for some reason?

Please post the version of the kernel you're using (`uname -a`) and run these commands from a fresh boot to try and get more information from `/dev/video0`:

    sudo modprobe v4l2loopback
    v4l2-ctl --all

bluezio, 2015-05-17:
Closing due to inactivity. Please reopen if you'd like to continue to look into this.

closed 2015-05-17 by bluezio

igitur, 2015-08-22:
Can we reopen this? I'm a different guy, but get exactly the same problem. I'm on Fedora 22, and just compiled and install a fresh v4l2loopback from source (v0.9.1). My output of `v4l2-ctl --all` is:
```
Driver Info (not using libv4l2):
	Driver name   : v4l2 loopback
	Card type     : Dummy video device (0x0000)
	Bus info      : platform:v4l2loopback-000
	Driver version: 4.1.4
	Capabilities  : 0x85208002
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x85208002
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
Priority: 0
Video output: 0 (loopback in)
Format Video Output:
	Width/Height  : 0/0
	Pixel Format  : 'BGR4'
	Field         : None
	Bytes per Line: 0
	Size Image    : 0
	Colorspace    : SRGB
	Flags         : 
Streaming Parameters Video Capture:
	Frames per second: 30.000 (30/1)
	Read buffers     : 8
Streaming Parameters Video Output:
	Frames per second: 30.000 (30/1)
	Write buffers    : 8

User Controls

                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0
                    keep_format (bool)   : default=0 value=0
              sustain_framerate (bool)   : default=0 value=0
                        timeout (int)    : min=0 max=100000 step=1 default=0 value=0
               timeout_image_io (bool)   : default=0 value=0
```

bluezio, 2015-08-22:
Which error message are you getting?

igitur, 2015-08-25:
The error message I get is:
`Failed to get 0 in input enumeration for /dev/video0. (25 - Inappropriate ioctl for device)`

I went through the entire history of this thread and can pretty much replicate every step of it.

I suspect this issue might be related: https://github.com/umlaeute/v4l2loopback/issues/83

bluezio, 2015-08-25:
Did you try that command that one of the commenters tried out?

$ gst-launch-0.10 videotestsrc ! v4l2sink device=/dev/video0

I'd say this is pretty much a basic test for your installation. My script doesn't do much beyond acting as a compatibility shim between v4l2loopback and IPWebcam, I'm afraid. If there's a bug in vl42loopback, there's not much I can do :-(.