agarciadom/ipwebcam-gst #18: timestamping problem during script execution

testardo, 2015-05-27:
Hola, I'm using IP Webcamera and the software where I run the script is Linux Mint 17.1 Rebecca. During the execution of the script the terminal gives back this following error:

>Informazioni di debug aggiuntive:
gstbasesink.c(2875): gst_base_sink_is_too_late (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
There may be a timestamping problem, or this computer is too slow.

How to resolve it?

bluezio, 2015-05-27:
Are you using a wifi connection, or a USB cable? Wifi connections can be
very slow.

testardo, 2015-05-27:
Hi bluezio,
thanks for the reply.
I did this operation without installing the android sdk. So this morning I installed it using this documentation: http://androidforums.com/threads/how-to-install-android-sdk-on-ubuntu.608406/.
I followed all the guide until to this line:

>Watch your phone's screen at this time. If it's not on, turn it on and look for an RSA popup which authorizes your device to your PC. When you see the popup, check the box to "always allow", then hit ok.

I plugged my phone with the usb cable, gave the ./adb devices command and finally I see my device in attachment, but when I restart your script the window gives to me this sentence:

>adb is not available: you'll have to use Wi-Fi, which will be slower. Next time, please install the Android SDK from developer.android.com/sdk.

and after the pavucontrol window etc. the terminal showes me again the timestamp problem.

bluezio, 2015-05-28:
If you're using ./adb to start ADB, it won't be part of your $PATH. It's
not really a problem, but in that case you'll need to customize ADB_PATH on
line 99 with the full absolute path to your adb binary.

testardo, 2015-05-29:
Hi bluezio,
thanks for the reply.
I configured the adb tool using this guide:
https://androidonlinux.wordpress.com/2013/05/12/setting-up-adb-on-linux/
and when I type the <b>./adb start-server</b> the terminal says that the daemon's starting on the port <b>5037</b>.
So on line 114 of your script I changed the number of the port then at on line 99, the default path with this line:
>/programmi/android-sdk-linux/platform-tools/adb

because the path where's the adb tool is found, it's in this path:

>/home/angelo/programmi/android-sdk-linux/platform-tools/

Then, I saved it and I restarted the script.

But after the <b>./prepare-videochat.sh</b> command, the script showes me this:
>angelo@EGOTRON ~/programmi/ipwebcam-gst-master $ ./prepare-videochat.sh
error: internal error

bluezio, 2015-05-30:
I'm afraid that doesn't give me enough information to debug the problem. Could you try adding a few "echo X" lines to the script, to find out where it's crashing?

testardo, 2015-06-04:
Hi bluezio,
thanks for the reply.
Actually we are changing the Internet Service Provider, so I don't have my internet connection.
I'll try it as soon as possible.

bluezio, 2015-07-02:
Closed due to inactivity. Please feel free to reopen when you get the time to try this again :-).

closed 2015-07-02 by bluezio