agarciadom/ipwebcam-gst #20: Internal data flow error

cjac, 2016-01-11:
Hey folks,

I get this error when running the prepare-videochat.sh script.  I'm on Debian Jessie.  I've got a newer version of the gstreamer plugins than you've tested with.

# Last tested with:
# - gstreamer-1.0-plugins-good 1.4.4-2
# - souphttpsrc version (included with plugins-good)
# - v4l2sink version (included with plugins-good)
# - v4l2loopback version 0.8.0-4

<pre>
$ sudo modinfo v4l2loopback
filename:       /lib/modules/3.16.0-4-amd64/updates/dkms/v4l2loopback.ko
license:        GPL
author:         Vasily Levin, IOhannes m zmoelnig <zmoelnig@iem.at>,Stefan Diewald,Anton Novikov
description:    V4L2 loopback video device
depends:        videodev
vermagic:       3.16.0-4-amd64 SMP mod_unload modversions
parm:           debug:debugging level (higher values == more verbose) (int)
parm:           max_buffers:how many buffers should be allocated (int)
parm:           max_openers:how many users can open loopback device (int)
parm:           devices:how many devices should be created (int)
parm:           video_nr:video device numbers (-1=auto, 0=/dev/video0, etc.) (array of int)
parm:           card_label:card labels for every device (array of charp)
parm:           exclusive_caps:whether to announce OUTPUT/CAPTURE capabilities exclusively or not (array of bool)
parm:           max_width:maximum frame width (int)
parm:           max_height:maximum frame height (int)
cjac@hype0:/usr/src/git/ipwebcam-gst$ dpkg -l | grep v4l2loop
ii  v4l2loopback-dkms                        0.8.0-4                              all          Source for the v4l2loopback driver (DKMS)
ii  v4l2loopback-utils                       0.8.0-4                              amd64        Commandline utilities for the for the v4l2-loopback module
cjac@hype0:/usr/src/git/ipwebcam-gst$ dpkg -l | grep gstreamer.\*good
ii  <b>gstreamer0.10-plugins-good:amd64         0.10.31-3+nmu4+b1</b>                    amd64        GStreamer plugins from the "good" set
ii  gstreamer0.10-plugins-good-doc           0.10.31-3+nmu4                       all          GStreamer documentation for plugins from the "good" set
ii  <b>gstreamer1.0-plugins-good:amd64          1.4.4-2</b>                              amd64        GStreamer plugins from the "good" set
ii  gstreamer1.0-plugins-good-doc            1.4.4-2                              all          GStreamer documentation for plugins from the "good" set

$ ./prepare-videochat.sh
/usr/bin/adb
5566
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
FOUND TAG      : found by element "wavparse0".
audio codec: Uncompressed 16-bit PCM audio
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:src: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:sink: caps = audio/x-raw-int, endianness=(int)1234, channels=(int)1, width=(int)16, depth=(int)16, signed=(boolean)true, rate=(int)44100
/GstPipeline:pipeline0/GstVolume:volume0.GstPad:src: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstVolume:volume0.GstPad:sink: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstRgLimiter:rglimiter0.GstPad:src: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstRgLimiter:rglimiter0.GstPad:sink: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstPulseSink:pulsesink0.GstPad:sink: caps = audio/x-raw-float, endianness=(int)1234, width=(int)32, rate=(int)44100, channels=(int)1
<b>ERROR</b>: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: <b>Internal data flow error.</b>
Additional debug info:
gstbasesrc.c(2625): gst_base_src_loop (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
streaming task paused, reason not-linked (-1)
Execution ended after 147280473 ns.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
/GstPipeline:pipeline0/GstPulseSink:pulsesink0.GstPad:sink: caps = NULL
/GstPipeline:pipeline0/GstRgLimiter:rglimiter0.GstPad:src: caps = NULL
/GstPipeline:pipeline0/GstRgLimiter:rglimiter0.GstPad:sink: caps = NULL
/GstPipeline:pipeline0/GstVolume:volume0.GstPad:src: caps = NULL
/GstPipeline:pipeline0/GstVolume:volume0.GstPad:sink: caps = NULL
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:src: caps = NULL
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:sink: caps = NULL
/GstPipeline:pipeline0/GstWavParse:wavparse0.GstPad:src: caps = NULL
/GstPipeline:pipeline0/GstMultipartDemux:multipartdemux0.GstPad:src_0: caps = NULL
Setting pipeline to NULL ...
Freeing pipeline ...
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
</pre>

bluezio, 2016-01-12:
I don't have any Debian Jessie systems to test with, but my first instinct would be to get Wireshark and have a look at which HTTP requests are being generated and if they're getting any 4xx/5xx replies. We'd need more information to know.

cjac, 2016-01-13:
Thanks for your reply.  It looks like it was a problem with the gst-launcher pipeline.  Some of the caps didn't line up.  Take a look at the patches I made to the prepare-videochat.sh script for details.

closed 2016-01-13 by cjac