agarciadom/ipwebcam-gst #22: Failure: Module initialization failed

sasindumendis, 2016-01-29:
I get this error when I try to run `./prepare-videochat.sh`.

I am on a Ubuntu 14.04 and trying to make use of an old W8 running Android 2.1 as a webcam.

Highly appreciate any guidance on troubleshooting this.

bluezio, 2016-01-30:
The SoupHTTPSrc is complaining that it cannot resolve the hostname:

    ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Cannot resolve hostname ()

Which IP and port are you using for the command provided? $IP and $PORT are variables that you should replace with the IP and port that IP Webcam is listening on: you should see them in the app once you start recording.

sasindumendis, 2016-01-30:
Yes I have set them in the Configuration section of the script like so:
```
# IP used by the phone in your wireless network
WIFI_IP=192.168.1.3

# Port on which IP Webcam is listening
PORT=8080
```

bluezio, 2016-01-30:
How are you running the command? Remember that if you simply paste the command in the console, the console does not have access to those variables in the script.

Make sure you run "export IP=192.168.1.3", then run "export PORT=8080" and then finally run the suggested command.

sasindumendis, 2016-01-30:
My bad, I tried again the command replacing `$IP:$PORT` with `192.168.1.3:8080` and got the video in a new window. But when I run the script I still get `Failure: Module initialization failed`.

bluezio, 2016-01-30:
Right, so the connection to IP Webcam works. Did you try troubleshooting step 1, which tests if v4l2loopback is available?

sasindumendis, 2016-01-30:
[image]

bluezio, 2016-01-30:
That means that v4l2loopback, IP Webcam and your wifi connection work. That probably means the problem lies with the PulseAudio modules and not with the kernel modules. Could you try running these two commands?

    pactl load-module module-null-sink
    pactl load-module module-echo-cancel

sasindumendis, 2016-01-31:
```
$ → pactl load-module module-null-sink
23

$ → pactl load-module module-echo-cancel
24
```

bluezio, 2016-02-01:
Yes, I was just checking if you had those two modules.

I'd probably start checking in which part of the script you get the error, by adding the appropriate "echo something" commands. I guess it'll probably be in the main "$GSTLAUNCH" command. If that's the case, try removing the audio side, for instance:

    "$GSTLAUNCH" -e -vt --gst-plugin-spew \
                 --gst-debug="$GST_DEBUG" \
      souphttpsrc location="$VIDEO_URL" do-timestamp=true is-live=true \
        ! multipartdemux \
        ! jpegdec \
        $GST_FLIP \
        ! $GST_VIDEO_CONVERTER \
        ! videoscale \
        ! videorate \
        ! $GST_VIDEO_CAPS \
        ! v4l2sink device="$DEVICE" sync=true \

You might also want to tweak the GST_DEBUG variable to increase the debug level of some of the GStreamer modules.

sasindumendis, 2016-02-01:
I really appreciate your continuous guidance on this :smile: 

Like you suggested I tried to isolate where exactly within the script error occurs by adding echo statements. This is what is found:

```
# echo "something" works here and above
if [ -z $SINK_ID ] ; then
    # echo "something" doesn't work here
    SINK_ID=$(pactl load-module module-null-sink \
                    sink_name="$SINK_NAME" \
                    $PA_AUDIO_CAPS \
                    sink_properties="device.description='IP\ Webcam'")
fi

if [ -z $ECANCEL_ID ] ; then
   # echo "something" works here
    ECANCEL_ID=$(pactl load-module module-echo-cancel \
                       sink_name="${SINK_NAME}_echo_cancel" \
                       source_master="$SINK_NAME.monitor" \
                       sink_master="$DEFAULT_SINK" \
                       $PA_AUDIO_CAPS \
                       aec_method="webrtc" save_aec=true use_volume_sharing=true)
    # echo "something" doesn't work here
fi
# echo "something" doesn't work here or further down
```

Removing the audio side from main `$GSTLAUNCH` command made no difference.

Tweaking the `GST_DEBUG` variable to increase the debug level of the GStreamer modules revealed no further helpful information (even when all of them are set to the max value: 5).

bluezio, 2016-02-02:
The module-echo-cancel command was added recently and it's nice but not strictly required, I think. Could you try removing it and seeing if that was the problem?

sasindumendis, 2016-02-02:
Yes, it did the trick! :)
Now I can get the video feed into Skype etc.

However I noticed my browser (Google Chrome) doesn't recognize it. For example if I try to use a browser based video chat app like http://appear.in or http://talky.io Chrome says not camera device avaiable.

When the script was running it threw a series of following error:

```
(gst-launch-1.0:5341): GStreamer-CRITICAL **: gst_segment_to_running_time: assertion 'segment->format == format' failed
```

Anyway really appreciate your support all the way. Thank you very much!

bluezio, 2016-02-02:
Great! No problem at all. Closing this issue then: there's not much that we can do about Chrome being finicky, I'm afraid :-). I'll make the `echo-cancel` module optional then, in case someone runs into the same issue.

closed 2016-02-02 by bluezio

sasindumendis, 2016-02-02:
Yeah I understand getting it to work on Chrome is beyond the scope :smile:  Just wanted to report everything I noticed so that you can further optimize the script. Thanks again!

lorefield, 2016-03-29:
I have this issue, but for me two modules will not load. One being the $ECANCEL_ID. After commenting-out the ECANCEL block, running the script still shows one "Failure: Module initialization failed". Playing with echos shows the error must be from the $SINK_ID block.

bluezio, 2016-03-30:
@lorefield, this issue is closed. Could you open a new one, so we can track this separately?