agarciadom/ipwebcam-gst #26: Double "Failure: Module initialization failed"

lorefield, 2016-03-30:
Running prepare-videochat.sh gives two occurrences of "Failure: Module initialization failed" and a "Failure: No such entity".

```console
./prepare-videochat.sh
Failure: Module initialization failed
Failure: Module initialization failed
Failure: No such entity
Press enter to end stream
```

Adding echo statements to the script locates the errors:

```bash
DEFAULT_SINK=$(pacmd dump | mawk '/set-default-sink/ {print $2}')
DEFAULT_SOURCE=$(pacmd dump | mawk '/set-default-source/ {print $2}')

SINK_NAME="ipwebcam"
SINK_ID=$(modid_by_sinkname $SINK_NAME)
ECANCEL_ID=$(modid_by_sinkname "${SINK_NAME}_echo_cancel")

echo loading: null-sink module
if [ -z $SINK_ID ] ; then
    SINK_ID=$(pactl load-module module-null-sink \
                    sink_name="$SINK_NAME" \
                    $PA_AUDIO_CAPS \
                    sink_properties="device.description='IP\ Webcam'")
fi

echo loading: echo-cancel module
if [ -z $ECANCEL_ID ] ; then
    ECANCEL_ID=$(pactl load-module module-echo-cancel \
                       sink_name="${SINK_NAME}_echo_cancel" \
                       source_master="$SINK_NAME.monitor" \
                       sink_master="$DEFAULT_SINK" \
                       $PA_AUDIO_CAPS \
                       aec_method="webrtc" save_aec=true use_volume_sharing=true) || true
fi

echo executing: pact1 set-default-source $SINK_NAME.monitor
pactl set-default-source $SINK_NAME.monitor
echo doing more stuff after this...
```

Running prepare-videochat.sh after that gives:

```console
./prepare-videochat.sh 
loading: null-sink module
Failure: Module initialization failed
loading: echo-cancel module
Failure: Module initialization failed
executing: pact1 set-default-source ipwebcam.monitor
Failure: No such entity
doing more stuff after this...
Press enter to end stream
```

Furthermore exiting the stream with [enter]  gives two occurences of "You have to specify a module 
index or name".

bluezio, 2016-04-22:
I am very sorry for the delay: this month has been very busy for me. Looking at your output, I'd say that the problem are the audio caps - there might be something in our defaults that your system does not agree with. Which version of Ubuntu are you using? Show me the results of `pactl --version` as well.

felixsanz, 2016-04-24:
i get the samer error.

pactl --version
pactl 8.0
Compiled with libpulse 8.0.0
Linked with libpulse 8.0.0

bluezio, 2016-04-26:
Hmm, you have the same version of PulseAudio than I do, and the script works fine on my system. Did you try the troubleshooting steps that are mentioned in the `prepare-videochat.sh` file?

bluezio, 2016-06-09:
Closing due to inactivity. Please reopen if you wish to continue looking into this.

closed 2016-06-09 by bluezio

felixsanz, 2016-06-09:
Well.. what do you want us to do?

bluezio, 2016-06-09:
Did you try the troubleshooting steps I mentioned?