agarciadom/ipwebcam-gst #31: can_run function working wrong in Ubuntu

laptander, 2016-06-21:
I was trying to run this script in Ubuntu (also in Linux Mint) and have found out that can_run function is working incorrectly. Take a look at this code:
```
#!/bin/bash
can_run() {
    (test -x "$1" || which "$1") &>/dev/null
}

if can_run date; then
	echo I can date
else
	echo I cannot date
fi

if can_run random_bull; then
	echo I can random_bull
else
	echo I cannot random_bull
fi
```
While on Arch Linux this code is going correctly (can date, cannot random_bull), in Ubuntu it always says that it can run everything (can date, can random_bull).
Also in Ubuntu script is not just end, but there is typed a path to binary that which command has found (not working redirection to /dev/null?) and new line in terminal after last echo (waking up from background?).

What I think to do:
1) remove test -x "$1", because it is useless. It only searches in current directory, and our binarias are not there.
2)  I thought that round brackets were made for preventing exit in case which status is not 0, because another shell will not have set -e installed. But this is not working, so maybe remove brackets?
3) Also can you please explain what ampersand means here? I thought that it is for backgrounding another shell, but am not sure in it. When I insert space between '&' and '>', it is not working even in Arch Linux. Do we need this ampersand?
4) All in all, I suggest just replace can_run function body with
`which "$1" >/dev/null` or even `command -v "$1" >/dev/null` because it bacomes working fine in Ubuntu.

bluezio, 2016-06-21:
Hi Laptander,

It's odd, since this has never been an issue for me.

1. Indeed, `test -x` only tries the current folder - perhaps it could be removed.

2. The round brackets are just there to group both things so we get their grouped return code.

3. `&>` is a Bash notation for redirect [both stdout and stderr at the same time](http://wiki.bash-hackers.org/syntax/redirection#redirecting_output_and_error_output). However, it now seems to be [deprecated](http://wiki.bash-hackers.org/scripting/obsolete). It could be a good idea to replace with `>/dev/null 2>/dev/null`.

4. Let's go for `which`: to be honest, I had never known about `command` until today, and on my Ubuntu system it doesn't have a manpage for some reason.

bluezio, 2016-06-21:
Changed in 1874570e59e68910016a73641fdb99bfcd914984. Please try it out.

laptander, 2016-06-21:
Perfect. The same issue in has_kernel_module() function
https://github.com/bluezio/ipwebcam-gst/blob/master/prepare-videochat.sh#L148

bluezio, 2016-06-21:
Done as well. Shall we mark as fixed?

laptander, 2016-06-21:
I think yes, thank you!

closed 2016-06-21 by laptander