agarciadom/ipwebcam-gst #33: multiple cams with wifi

lost-hope, 2016-07-11:
Is it possible to use more then one phone as a Webcam over WIFI?

bluezio, 2016-07-11:
I've never tried this and it's not supported, as far as I know. If you get
it working, let me know! :-)

laptander, 2016-07-14:
@lost-hope Of course it is possible. But it is not realized yet (somebody need this?). I mean technically it will not not differ from registering first camera.
Maybe you ask if it will be stable picture while both cameras are transmitting information via wifi?

lost-hope, 2016-07-14:
I have tried to get the first smartphone to /dev/video1. this worked. but when I copied the script and changed the ip adress but this didn't worked
where can I specify which device is going to be used

bluezio, 2016-07-15:
Right, the problem is in lines 312-319. These lines detect the first v4l2loopback device and use that.

What we could do is make it so if you run the script with DEVICE=/dev/video2 ./prepare-videochat.sh, it will override this autodetection and use that. You'll have to load the v4l2loopback module in advance with the "devices=N" option, though.

I'll make the change and also add how to do it at the beginning of the script.

bluezio, 2016-07-15:
I've pushed some changes that should make it possible to have two copies of the script using different WIFI_IP and DEVICE settings. The line for DEVICE is commented by default so we keep the same autodetection as usual for new users.

See the new "MULTIPLE WEBCAMS" section in the script for details on how to do it. Try it out and let us know if it works for you :-).

bluezio, 2016-07-27:
Closing due to inactivity. Feel free to open a new issue if you run into trouble with this new feature.

closed 2016-07-27 by bluezio