agarciadom/ipwebcam-gst #34: getting '

sandeshc, 2016-07-26:


bluezio, 2016-07-27:
I'm not sure I understand the problem. Let me ask a few questions to clarify things:

* Does the script work the first time you run it from a freshly booted system, and fail the second time?
* Do you already have a webcam in your machine?
* Before you run the script for the first time, do you have any /dev/video* files?

sandeshc, 2016-07-27:
1. The script doesn't give any error and works fine first time after `sudo modprobe`
2. The webcam I have on my machine isn't being recognized (all applications report : "no video device")
3. Yes, `/dev/video0` exists before running the script for the first time

bluezio, 2016-07-27:
Does the script work fine on the first time without your manual `sudo modprobe`, or does it fail?

Essentially, the script sees if you have v4l2loopback running and if not, it loads the module for you. It then checks if you set DEVICE manually, and if you hadn't it uses v4l2-ctl to detect the first /dev/video* that comes from v4l2loopback (lines 390-403).

I suspect there might be something going wrong in this code on your system. I'd advise peppering the code with a few `echo` commands to see how it goes and if it's computing DEVICE correctly.

sandeshc, 2016-07-27:
After boot, I ran to script to find that `/dev/video1` created after `sudo modprobe v4l2loopback` had vanished. Thus the script by default took `/dev/video0` as the sink. The problem is that if this happens, the video that has been captured using `souphttpsrc` does not get dumped to `/dev/video0`.

On the other hand after `sudo modprobe v4l2loopback`, `/dev/video1` gets created and the script uses the same. This at least works for the first time after `modprobe`.

I had tried to debug the script, everything works fine until `v4l2sink device=/dev/video0` which seems to be the source of the error.

bluezio, 2016-07-27:
Interesting! So it seems that the bit that is failing is the "is v4l2loopback running" check. Try changing lines 362--368 to something like this:

```
if lsmod | grep v4l2loopback >/dev/null 2>/dev/null; then
    echo Module is already loaded
elif [ $CAPTURE_STREAM = v -o $CAPTURE_STREAM = av ]; then
    echo Loading module
    sudo modprobe v4l2loopback
fi
```

Try running the script from a fresh boot and let me know if it goes for "Module is already loaded" or for "Loading module". If it goes to "Module is already loaded" without v4l2loopback having being loaded, that'd point to `lsmod | grep v4l2loopback >/dev/null 2>/dev/null` being the issue here.

sandeshc, 2016-07-27:
Haven't done `sudo modprobe` after booting the machine. It does say `Module is already loaded'.

On running `lsmod | grep v4l2loopback`, got the following output

```
v4l2loopback_dc        23766  1 
videodev              134688  2 v4l2loopback_dc
```

bluezio, 2016-07-28:
Wait, that's odd: it's not "v4l2loopback" but "v4l2loopback_dc". Could you run `sudo modprobe v4l2loopback` and check `lsmod | grep v4l2loopback` again, to see if you'd have two `v4l2loopback` entries?

If you do, I wonder where that `_dc` entry is coming from. The fix would be simple (changing the code to use `lsmod | grep -w v4l2loopback`), but it's still strange.

sandeshc, 2016-07-28:
It seems that `v4l2loopback_dc` entry is being loaded on boot. After `sudo modprobe v4l2loopback` as expected got a new entry for `v4l2loopback`. The initial set of command are working fine. Could be helpful if the script was modified for `lsmod | grep -w v4l2loopback` !

Thanks for help and the fix :) Could create a PR if you'd like.

bluezio, 2016-07-28:
The PR would be welcome, thanks :-). Please make sure you put "refs #34" in
the commit message so it's related to this discussion.

sandeshc, 2016-07-29:
Fixed with PR #37

closed 2016-07-29 by sandeshc