agarciadom/ipwebcam-gst #35: Having issues connecting after daemon startup

DastardlyMut, 2016-07-27:
Hello! 

I'm having an issue with ./prepare-videochat.sh.

bluezio, 2016-07-27:
Hmm, that sounds more like an ADB issue than an issue from the script. First, are you using a USB cable, or a Wi-Fi connection? If you're using ADB, does `adb devices` work? It should list your device.

If you're using Wi-Fi, did you set up WIFI_IP correctly and did you try pinging the IP you put there?

Does something like this help?

http://stackoverflow.com/questions/21049253/adb-devices-return-error-protocol-fault-no-status

Using the script in combination with OpenCV sounds very interesting. If you get it working, I wouldn't mind adding a link to your work in the README :-D.

DastardlyMut, 2016-07-27:
I see! I'm trying to use the WIFI at the moment.
Unfortunately the USB didn't seem to be working so ADB wasn't picking it up (should probably get that working). This is why I was trying straight with the router. 

I have set up WIFI_IP in the script correctly from what I can tell. Have just tried pinging and it returns unknown host. This is strange tho because I can access the feed in my browser and with VLC's streaming.

Yes, of course, that's no problem :) Hopefully the problem can be solved! Would be awesome to be able to use my phone camera for image processing.

bluezio, 2016-07-27:
"Unknown host" sounds like a DNS problem, though. You probably want to solve that first before anything else - I can't help much with ADB issues, sorry :-/. Are you using a hostname instead of a fixed IP, then?

DastardlyMut, 2016-07-27:
For sure. 

It seems to be pinging correctly when I exclude the port of the IP address of the camera, so not quite sure why it wouldn't with the port included. Sorry, need to brush up on my networking a bit.

I'm using a fixed IP as far as I can tell. Literally, I have my phone connected to my router WIFI and my pc is connected to my router via ethernet. 

I will see if I can adb working!

bluezio, 2016-07-27:
Ah, that's probably it. `ping` only takes an IP: the port is already decided for you (it's the ICMP port), so if you put `:port` it thinks it's a name instead and that clearly doesn't go well :-).

DastardlyMut, 2016-07-27:
Aha, I understand! Well played! :) 

I will let you know if I can get the adb devices and that working too. I can't see it being too much of a hard thing to do.

DastardlyMut, 2016-07-27:
If the script works, am I able to reference the IP cam's stream as a plain video device? i.e Webcam?

bluezio, 2016-07-28:
It appears as a regular Linux webcam, so you should be able to access it through the Video 4 Linux (v4l) libraries as far as I know.

DastardlyMut, 2016-07-29:
Fantastic.

Sorry for late replies, have been busy past few days. Will work on it again this weekend.

DastardlyMut, 2016-08-04:
Have managed to get IP cam working with OpenCV (but without using the script) :D 

Turns out my OpenCV was not built correctly due to some missing dependencies. After installing the correct dependencies and rebuilding opencv I was able to easily capture video from the IPCam App on my phone using the line:
` vidcap = cv2.VideoCapture("http://10.0.0.4:8080/video") `

At this point, I'm not sure the script is necessary to use when using OpenCV. 

I still need to get it working 100% though. As mentioned, I think it's to do with my adb not working correctly.

bluezio, 2016-08-04:
Wow, That is indeed very convenient!

Perhaps you don't need the script after all for OpenCV. This is mostly to make desktop webcam apps work with the phone: if OpenCV understands MJPEG streams over HTTP directly, there's no need for GStreamer and no need for v4l2loopback.

bluezio, 2016-08-16:
Did you look into this, or should we close the bug for now?

bluezio, 2016-10-22:
Closing due to inactivity.

closed 2016-10-22 by bluezio