agarciadom/ipwebcam-gst #39: No Video, but ADB is working

blipton, 2016-08-14:
I'm using a beaglebone black (debian 8.4) and have installed the ipwebcam-gst script along with script found on the droidcam website.

So on the BBB I have installed : android-tools-adb , android-tools-fastboot , v4l2loopback-dkms, v4l2loopback, and v4l2loopback_dc. 

I verified the last 2 are listed in 'lsmod', and that /dev/video0 is present when I plug in.   Also,  'adb version' returns 'Android Debug Bridge version 1.0.31'.     However, when I start the webcam application (DroidCam, IP Webcam,  WO Webcam) on android (Galaxy S4), I don't get any video (ex. using Cheese on the beaglebone).

Interestingly, droidcam has an option to toggle the flash, and that works!  So the adb/usb communication is working, but, for some reason, not the video.

bluezio, 2016-08-16:
DroidCam is not supported, I'm afraid: we only support IP Webcam.

That said: I guess you checked with a browser if the setup worked, right? Did you try step 2 of the troubleshooting guide in the script? Also, you could try increasing the debug level of the GStreamer pipeline by editing the GST_DEBUG variable.

bluezio, 2016-10-22:
Closing due to inactivity.

closed 2016-10-22 by bluezio