agarciadom/ipwebcam-gst #42: GStreamer test video source OK; unable to test connection to phone/script. I'm stumped.

amsheppard, 2017-01-07:
Hello -

I'm trying to used IP Webcam as a USB connected camera.

# Troubleshooting Preformed Thus Far
When trying to execute:

`gst-launch-1.0 souphttpsrc location="http://192.168.0.147:8080/videofeed" \ do-timestamp=true is-live=true \ ! multipartdemux ! jpegdec ! videoconvert ! ximagesink`

It returns:
`WARNING: erroneous pipeline: syntax error`

Launching: 
`gst-launch-1.0 videotestsrc ! tee ! v4l2sink device=/dev/video1 & mplayer -tv device=/dev/video1 tv://`
returns the test card. (Note piping `tee` was found to be necessary based on a different issue which I no longer can recall the related source/origin).

v4l2loopback-dkms 0.9.1-4 module taints kernel; won't load if added to /etc/modules
using manual "modprobe v4l2loopback" method instead, which creates /dev/video1

When using the script, the `android.intent.action` does fire as expect but then 'bombs' per the output below.

I can only suspect v4l2loopback's issue of tainting the kernel may be the root issue vs the script but am still @ a loss as to why `gst-launch-1.0 souphttpsrc [...]` will not execute. Any assistance is appreciated.

## Using The Script

```
andrew@x220-eos:~/.bin$ ./prepare-videochat.sh 
Starting: Intent { act=android.intent.action.MAIN cmp=com.pas.webcam/.Rolling }
Failure: Module initialization failed
```
# Specs

## Host
Elementary OS 0.4 Loki Released, Based on Ubuntu 16.04
`Linux x220-eos 4.4.0-57-generic #78-Ubuntu SMP Fri Dec 9 23:50:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

```
root@x220-eos:~# ll /dev/video*
crw-rw----+ 1 andrew andrew 81, 0 Jan  7 03:29 /dev/video0
crw-rw----+ 1 andrew andrew 81, 1 Jan  7 03:31 /dev/video1

```
```
android-tools-adb                              5.1.1r36+git20160322-0ubuntu3                           amd64
gstreamer1.0-x:amd64                           1.8.2-1ubuntu0.2                                        amd64
gstreamer1.0-plugins-good:amd64                1.8.2-1ubuntu0.3                                        amd64
gstreamer0.10-tools                            0.10.36-1.5ubuntu1                                      amd64
v4l2loopback-dkms                              0.9.1-4                                                 all
```

## Phone
Phone is a single, stock LG G3 (D852) using PTP connectivity & is seen by ADB, connected via LG branded USB cable.
```
List of devices attached 
LGD8522683982e	device
```

```
Android version: 5.0.1
Kernel Version: 3.4.0
Build number: LRX21Y
Software version: D85220g
```

bluezio, 2017-01-09:
It isn't that surprising that the kernel would be "tainted": it's not an official Linux kernel module AFAIK. Regardless, I would treat it as a harmless warning - the problem must be somewhere else. I would suggest looking into dmesg for v4l2 messages, and I would also try simplifying the pipeline that is failing. For instance, try leaving only the first component and then keep adding things back:

    gst-launch-1.0 souphttpsrc location="http://192.168.0.147:8080/videofeed"

In fact, I just tried copying and pasting that into my command line and it seemed to work. Putting everything into one line also worked:

    gst-launch-1.0 souphttpsrc location="http://192.168.0.147:8080/videofeed" ! multipartdemux ! jpegdec ! videoconvert ! ximagesink

I think that perhaps you may have made a small mistake with the backslashes ("\"), and may have turned some of those "! x !" bits into "!x !" inadvertently, which would be interpreted by bash into the last command with "x" on it and will mangle the command to no end.

Make sure that there are spaces between the !, and be careful with those \: I would only put them right before a newline.

amsheppard, 2017-01-14:
Well, _that_ certainly made a difference! Thanks! I can get the `gst-launch-1.0 souphttpsrc [...]` to stream over wifi now. That's one down.

On a cold reboot of both the computer & the phone, I'm having some trouble w/ `prepare-videochat.sh` to execute. I've set it for video only, 30fps, the aforementioned $IP & $DEVICE .

IP=192.168.0.147
PORT=8080
WIDTH=1280
HEIGHT=720
GST_FPS=30
CAPTURE_STREAM=v
DEVICE=/dev/video1
[ confirmed same settings in app + quality of 100 in app]

Output gives `Loading module`, I enter root's pwd , then a Zenity output of "_adb is available, but the phone is not plugged in. Are you sure you want to use Wi-Fi (slower)? [...]_" . I choose _no_. 

$IP is shown on the phone's screen but _I am_ able to see the stream in VLC via http://localhost:8080/videofeed

... so it's odd that I'm getting that Zenity prompt, IMO.

Any advice how to deal w/ the ~2 sec lag between the camera & stream (via VLC streaming) or the ~1 sec delay w/ Javascript in fullscreen? Settings currently @ 75% but seem same even @ 50%.

My target res is 1080p, recording to the computer instead of a SD card. Should I expect a lag over USB even @ 720p?

bluezio, 2017-01-15:
Normally using USB makes things more manageable - it isn't lag free for me either, but I guess that's the price to pay for all the piping going on :-). Indeed, I wouldn't go for 1080p over USB 2.0. If it is falling back to Wi-Fi, you *will* get a lot of lag.

Now, if the phone is plugged in *but* you're getting that Zenity prompt, that would only mean that `adb get-state` is not returning "device". It is odd, since it is being listed from ADB, but perhaps the output format has changed slightly in current versions. Could you try running `adb get-state` while the phone is plugged in and picked up by ADB?

amsheppard, 2017-01-15:
Would 720p @ 30fps be a reasonable goal?

Same result if I wake the screen or not:

```
andrew@x220-eos:~$ adb devices

List of devices attached 
LGD8522683982e	device
```

```
andrew@x220-eos:~$ adb get-state
device
````

bluezio, 2017-01-15:
I think it should be reasonable, yes.

Now that I think about it, if http://localhost:8080 works then you are getting the right forwarding and USB is working fine. I think it's working for the most part - the Zenity prompt might be because the first time it checks "adb get-state" it might be reporting something different, and when you say "no" it tries again and it works as it should.

Now, the speed - with USB working, that's as good as it's going to get. I can't do much there, sorry :-(.

amsheppard, 2017-01-17:
> Now, the speed - with USB working, that's as good as it's going to get. I can't do much there, sorry :-(.

I refuse to believe that. ;-) I'm going to have some fun testing this over the next few days. You're amazing & I thank you.

amsheppard, 2017-01-17:
[Closing issue as primary issue resolved]

closed 2017-01-17 by amsheppard