agarciadom/ipwebcam-gst #48: Package 'gstreamer0.10-pulseaudio' has no installation candidate on Debian 9.3

anant2306, 2018-02-04:
Hi, i tried executing the script on Debian 9.3 but it is failing with below error:

"Package 'gstreamer0.10-pulseaudio' has no installation candidate"

This package is not available in debian repository.Also, an updated version is already installed on my system - gstreamer1.0-pulseaudio.

bluezio, 2018-02-04:
Can you try changing the script to gstreamer1.0-pulseaudio and reporting back?

Sorry, but I do not have an installation of Debian 9.3 around that I can try it on.

anant2306, 2018-02-05:
Hi, thanks for reply.

I did tried that yesterday and now getting different error - Failed to Initialize module.

I dont have a screenshot right now so couldn't post it.I will try attach the screenshot today eod.

Since you do not have any installation,  let know if i could be of any help in providing any further info.

⁣Regards,
Anant

Sent from Blue ​

0100001001000010, 2018-02-07:
I have this issue too. This makes the app completely useless to me.

bluezio, 2018-02-07:
That means the pipeline is somehow not starting properly. GStreamer 1.0 might have made some changes that require tweaking the components we use: this has happened a few times over the history of the script.

Could you attempt the troubleshooting steps that are documented further below in the script, and report back?

https://github.com/bluezio/ipwebcam-gst/blob/master/prepare-videochat.sh#L79

0100001001000010, 2018-02-08:
The `gst-launch` command does not exist on my system.

bluezio, 2018-02-15:
Recent versions of Debian and Ubuntu use `gst-launch-1.0` instead. Try it out and let me know how it goes :-).

bluezio, 2018-02-24:
No response from the user: closing for now until further input.

closed 2018-02-24 by bluezio