agarciadom/ipwebcam-gst #56: Could not install the v4l2loopback kernel module through apt-get [on Parrot OS (debian)]

poquex, 2019-08-15:
Im kinda noob on linux so, maybe it is a really easy to fix problem but, I really don't have a clue.

bluezio, 2019-08-16:
Did the `v4l2loopback-dkms` package install correctly? Check the packages listed in your system, like this:

```
$ dpkg -l | grep v4l2loopback-dkms
ii  v4l2loopback-dkms                                           0.12.1-1                                    all          Source for the v4l2loopback driver (DKMS)
```

bluezio, 2020-02-06:
Closing due to inactivity.

closed 2020-02-06 by bluezio

banjoman72, 2020-04-08:
Hi Bluzio, I have the same prob using ubuntu bionic.
I ran the dpkg command and it cameback +ve but the script still complains

<pre><font color="#729FCF"><b>~/ipwebcam-gst-master</b></font>$ ./prepare-videochat.sh 
find: ‘/lib/modules/5.3.0-46-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-46-generic/extramodules’: No such file or directory
Trying to install v4l2loopback-dkms package.
[sudo] password for ben: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l2loopback-dkms is already the newest version (0.10.0-1ubuntu1.2)</pre>

Is Ubuntu organising the files a slightly different way than expected?
What can I do next?
Thanks in advance :)

banjoman72, 2020-04-08:
Sorry I only read the comments in the script after posting. 
<pre><font color="#8AE234"><b>ben@dwarf99</b></font>:<font color="#729FCF"><b>~</b></font>$ sudo modprobe -r v4l2loopback
<font color="#8AE234"><b>ben@dwarf99</b></font>:<font color="#729FCF"><b>~</b></font>$ ls /dev/video*
<span style="background-color:#2E3436"><font color="#FCE94F"><b>/dev/video0</b></font></span>  <span style="background-color:#2E3436"><font color="#FCE94F"><b>/dev/video1</b></font></span>
<font color="#8AE234"><b>ben@dwarf99</b></font>:<font color="#729FCF"><b>~</b></font>$ sudo modprobe  v4l2loopback
<font color="#8AE234"><b>ben@dwarf99</b></font>:<font color="#729FCF"><b>~</b></font>$ ls /dev/video*
<span style="background-color:#2E3436"><font color="#FCE94F"><b>/dev/video0</b></font></span>  <span style="background-color:#2E3436"><font color="#FCE94F"><b>/dev/video1</b></font></span>  <span style="background-color:#2E3436"><font color="#FCE94F"><b>/dev/video2</b></font></span>
<font color="#8AE234"><b>ben@dwarf99</b></font>:<font color="#729FCF"><b>~</b></font>$ v4l2-ctl -D -d /dev/video2
Driver Info (not using libv4l2):
	Driver name   : v4l2 loopback
	Card type     : Dummy video device (0x0000)
	Bus info      : platform:v4l2loopback-000
	Driver version: 5.3.18
	Capabilities  : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities
	Device Caps   : 0x85208003
		Video Capture
		Video Output
		Video Memory-to-Memory
		Read/Write
		Streaming
		Extended Pix Format
		Device Capabilities</pre>

so the phone is recognised, but it isn't using the libv4l2 driver? I hope this helps

banjoman72, 2020-04-10:
It now works!!
solution: install linux-generic

**code**: sudo apt install linux-generic

**attribution**: user: HD1080ide **thread:** https://github.com/umlaeute/v4l2loopback/issues/59

jcmguerrero, 2020-04-15:
Hi,
I've been reading the feed and following  the steps, with the same results. But when I try 
code: sudo apt install linux-generic

I get the message:
linux-generic ya está en su versión más reciente (5.3.0.46.39).
(in english [more or less] linux-generic is installed and it has the mos recent version (5.3.0.46.39)  (please, forgive my poor english)

Any ideas?
Thanks in advance!!!

jcmguerrero, 2020-04-15:
> Hi,
> I've been reading the feed and following the steps, with the same results. But when I try
> code: sudo apt install linux-generic
>
> I get the message:
> linux-generic ya está en su versión más [...]

Sorry, it works!!
Thanks for all your previous ideas!!

jcmguerrero, 2020-04-15:
Sorry, it's me again...
I've tried VLC link but when I open video capture devide I get this error:

Su entrada no puede abrirse:
VLC es incapaz de abrir el MRL «v4l2:///dev/video2». Vea el registro para más detalles.

So, I don't know what can I do. Can you help me?

Thanks!!

bluezio, 2020-04-15:
What does `ls -l /dev/video2` show, and what does VLC show?

jcmguerrero, 2020-04-15:
Thanks **bluzio**, I tried DroidCam and worked fine. Anyway `ls -l /dev/video2` shows:

crw-rw----+ 1 root video 81, 2 abr 15 12:30 /dev/video2

And VLC:
VLC es incapaz de abrir el MRL «v4l2:///dev/video2». Vea el registro para más detalles.

Thanks again

bluezio, 2020-04-15:
Try this command, it should print "OK" if your permissions are good:

```
test -r /dev/video2 && test -w /dev/video2 && echo OK || echo NOT_OK
```

For VLC, what does the VLC log say?

nmolanog, 2020-04-28:
I have this issue: while running "dpkg -l | grep v4l2loopback-dkms" i get:

`find: ‘/lib/modules/5.3.0-46-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-46-generic/extramodules’: No such file or directory
Trying to install v4l2loopback-dkms package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l2loopback-dkms is already the newest version (0.10.0-1ubuntu1.2).
The following packages were automatically installed and are no longer required:
  efibootmgr gir1.2-appindicator3-0.1 gir1.2-geocodeglib-1.0 libfwup1 libllvm8 libllvm8:i386
  redshift ubuntu-web-launchers
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
find: ‘/lib/modules/5.3.0-46-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-46-generic/extramodules’: No such file or directory`

I also got a window with the message: "Could not install the v4l2loopback kernel module through apt-get."

output for `dpkg -l | grep v4l2loopback-dkms`:

`ii  v4l2loopback-dkms                          0.10.0-1ubuntu1.2                                all          Source for the v4l2loopback driver (DKMS)
`

bluezio, 2020-04-28:
@nmolanog could you open a different issue for your problem?

nmolanog, 2020-04-28:
> @nmolanog could you open a different issue for your problem?

done