agarciadom/ipwebcam-gst #59: ipwebcam client NOT starting

dosgeek95, 2020-02-05:
So, I have used this program in the past with no issues and just recently got back into it. Currently using it on Lubuntu 18.04 (minimal GUI) which is fully up-to-date and also have Android SDK installed through apt-get.

Unfortunately, I could NOT get the program started as I got stuck on this particular error:

"Could not install the v4l2loopback kernel module through apt-get."

But here's the thing. v4l2loopback appears to be installed properly on the system as shown below:

$ dpkg -l | grep v4l2loopback-dkms
ii  v4l2loopback-dkms                     0.10.0-1ubuntu1.2                               all          Source for the v4l2loopback driver (DKMS)

AND the module file for v4l2loopback DOES exist in the following directory as shown below, which is one of the three that prepare_videochat.sh searches in according to the source code.

$ find /lib/modules/$(uname -r)/extra -name "v4l2loopback.ko"
/lib/modules/4.15.0-76-generic/extra/v4l2loopback.ko

So, any input on this would be greatly appreciated as I'm completely dumbfounded as to what this issue is.

bluezio, 2020-02-06:
Could you try this? Change:

```
 if [ 
	`find /lib/modules/$(uname -r)/ -name "$MODULE.ko.*" | egrep '.*' || 
	find /lib/modules/$(uname -r)/extra -name "$MODULE.ko.*" | egrep '.*'||
	find /lib/modules/$(uname -r)/extramodules -name "$MODULE.ko.*" | egrep '.*'` ]; then
        return 0
       else
        return 1
       fi
```

to:

```
 if [ 
	`find /lib/modules/$(uname -r)/ -name "$MODULE.ko*" | egrep '.*' || 
	find /lib/modules/$(uname -r)/extra -name "$MODULE.ko*" | egrep '.*'||
	find /lib/modules/$(uname -r)/extramodules -name "$MODULE.ko*" | egrep '.*'` ]; then
        return 0
       else
        return 1
       fi
```

I don't know if that ".*" at the end of `-name "$MODULE.ko.*"` might be causing issues. It hasn't been a problem before, but who knows!

bluezio, 2020-02-12:
Did you try this change?

dosgeek95, 2020-02-12:
Heyo. Sorry about the delay in response. Had other things in mind.

Anyhow, I was gonna try the fix that you mentioned, but before I was able to apply it, the program suddenly started working again. ....Huh?

Because previously it complained about v4l2loopback not existing in the Linux module folders when it did, and NOW it starts to work. 

Now that I think about it, I have been restarting the computer in question while I was doing other things to it. Maybe it had something to do with dependencies not initializing at full right after install, but I'm not really sure.

bluezio, 2020-02-13:
Huh, that's strange. I'll close the issue for now, then.

closed 2020-02-13 by bluezio