agarciadom/ipwebcam-gst #61: Script Crash

cigarplug, 2020-03-21:
Hi
When I execute the script, it crashes within seconds:

```
[moocow@goat ipwebcam-gst]$ ./prepare-videochat.sh 
Starting: Intent { act=android.intent.action.MAIN cmp=com.pas.webcam/.Rolling }
Security exception: Permission Denial: starting Intent { act=android.intent.action.MAIN flg=0x10000000 cmp=com.pas.webcam/.Rolling } from null (pid=21852, uid=2000) not exported from uid 10070

java.lang.SecurityException: Permission Denial: starting Intent { act=android.intent.action.MAIN flg=0x10000000 cmp=com.pas.webcam/.Rolling } from null (pid=21852, uid=2000) not exported from uid 10070
	at com.android.server.wm.ActivityStackSupervisor.checkStartAnyActivityPermission(ActivityStackSupervisor.java:1361)
	at com.android.server.wm.ActivityStarter.startActivity(ActivityStarter.java:896)
	at com.android.server.wm.ActivityStarter.startActivity(ActivityStarter.java:676)
	at com.android.server.wm.ActivityStarter.startActivityMayWait(ActivityStarter.java:1776)
	at com.android.server.wm.ActivityStarter.execute(ActivityStarter.java:607)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1619)
	at com.android.server.wm.ActivityTaskManagerService.startActivityAsUser(ActivityTaskManagerService.java:1516)
	at com.android.server.am.ActivityManagerService.startActivityAsUser(ActivityManagerService.java:4191)
	at com.android.server.am.ActivityManagerShellCommand.runStartActivity(ActivityManagerShellCommand.java:518)
	at com.android.server.am.ActivityManagerShellCommand.onCommand(ActivityManagerShellCommand.java:172)
	at android.os.ShellCommand.exec(ShellCommand.java:104)
	at com.android.server.am.ActivityManagerService.onShellCommand(ActivityManagerService.java:11809)
	at android.os.Binder.shellCommand(Binder.java:881)
	at android.os.Binder.onTransact(Binder.java:765)
	at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:5201)
	at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:3356)
	at android.os.Binder.execTransactInternal(Binder.java:1021)
	at android.os.Binder.execTransact(Binder.java:994)
```

Is there something I could be doing wrong? Missing dependencies?
I am using Arch Linux:

```
[moocow@goat ipwebcam-gst]$ uname -a
Linux goat 5.5.10-arch1-1 #1 SMP PREEMPT Wed, 18 Mar 2020 08:40:35 +0000 x86_64 GNU/Linux
```

luco5826, 2020-03-23:
Same problem here, running Debian 10, kernel 4.19.0-9
Works flawlessly with the command above but on teams (based on skype) the Dummy video device returns the same image.
`v4l2l-ctl --list-device` correctly shows the loop back device:
```
Dummy video device (0x0000) (platform:v4l2loopback-000):
	/dev/video2
```
I love the idea of this app, so I'd be glad to help if possible

bluezio, 2020-03-24:
What is the error output you get from the script?

luco5826, 2020-03-24:
On my machine, nothing from the shell but a dialog appear like this 

[image]

And up until yesterday it was showing but not working. Right now I re-did it and not it works I did not change anything, I just shut down my PC and turned it back on today. 
Maybe could be the re-boot that fixed the issue? I saw another issue noticing the same "fix"

cigarplug, 2020-03-24:
Rebooting the system fixed it for me too. Thanks for your work, much appreciate. <3

closed 2020-03-24 by cigarplug