agarciadom/ipwebcam-gst #63: Configuration of v4l2loopback failes on 5.3.0-43-generic kernel

wluberti, 2020-03-25:
I cloned this repo and executed `bash ./prepare-videochat.sh`. After filling in my sudo password the installation of the packaged succeed but during the configuration I get on pop-up error `Could not install the v4l2loopback kernel module through apt-get.`

stacktrace:
```
bash ./prepare-videochat.sh                                                                             3298  09:05:22  
find: ‘/lib/modules/5.3.0-43-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-43-generic/extramodules’: No such file or directory
Trying to install v4l2loopback-dkms package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l2loopback-dkms is already the newest version (0.12.2-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
find: ‘/lib/modules/5.3.0-43-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-43-generic/extramodules’: No such file or directory
```
(i've create the two missing directories, but the problem persists)

The package seems to be installed:
```
dpkg -l | grep v4l2loopback-dkms
ii    v4l2loopback-dkms    0.12.2-1    all    Source for the v4l2loopback driver (DKMS)
```

Any ideas?

Jakaria08, 2020-03-26:
Facing the same problem on Ubuntu 18.04

luco5826, 2020-03-26:
Hi!
Had the same problem on Debian 10
Try running
```
sudo modprobe v4l2loopback
```
It will force load the v4l2loopback module.
Then run again the script

wluberti, 2020-03-26:
The script ran after doing `sudo modprobe v4l2loopback`. However a pop-up came with `adb is not available: you'll have to use Wi-Fi, which will be slower. Next time, please install the Android SDK from developer.android.com/sdk or install adb package in Ubuntu`. I've edited the .sh file for my IP address and it says the audio and video feed is available, but the webcam app (cheese) can't find the streams. 

Will have a look again tomorrow. Thanks so far!

Temetias, 2020-04-01:
> The script ran after doing `sudo modprobe v4l2loopback`. However a pop-up came with `adb is not available: you'll have to use Wi-Fi, which will be slower. Next time, please install the Android SDK [...]

Hah, I'm actually stuck on the same exact spot with the same steps. Did you have any luck fixing it?

I did notice that running cheese from terminal does give some assertion errors like CHEESE_IS_CAMERA_DEVICE failed and such which does suggest that the camera not really found even.

gpatel-fr, 2020-04-03:
Hello
just to share what I did under Ubuntu 18.04

```
 # check if the user has the pulse gst plugin installed
@@ -557,6 +558,7 @@
     ! videoscale \
     ! videorate \
     ! $GST_VIDEO_CAPS \
+    ! identity drop-allocation=1 \
     ! v4l2sink device="$DEVICE" sync=$SYNC
 }
```

 
then use updated version of v4l2loopback, see:
https://github.com/ShayBox/Mon2Cam/issues/8
without this update, v4l-info returns an error for video_capture, so:

sudo apt install v4l2loopback-utils
sudo rmmod v4l2loopback
sudo dkms uninstall -m v4l2loopback -v 0.10.0
sudo mkdir /usr/src/v4l2loopback-0.12.3
(copy v4l2loopback.c and v4l2loopback_formats.h  file from new github version, copy dkms.conf and Makefile from v4l2loopback-0.10.0, adapt dkms.conf to set package version to 0.12.3)
sudo dkms add -m v4l2loopback -v 0.12.3
sudo dkms build -m v4l2loopback -v 0.12.3
sudo dkms install -m v4l2loopback -v 0.12.3
sudo modprobe v4l2loopback exclusive_caps=1 max_buffers=2

sudo v4l2loopback-ctl set-fps 14 /dev/video0

it's also necessary for each run to unload adb (killall adb)

with this, I was able to use vlc, cheese, ffplay and chromium. It's slow and very unstable but possibly it can come from the phone, an antique low-cost Android 4.0 from 2013 (I have nothing better to use as an additional webcam ATM - that's why I tried to limit frame rate)

Hope this helps someone.

bluezio, 2020-04-14:
Thanks for this! I'll leave this issue open for now so others can refer to it. Hopefully `v4l2loopback-dkms` will be updated to 0.12.3 soon and we will be able to close this issue :-).

sorachan, 2020-04-17:
I ended up circumventing the problem by not using the script / GStreamer. Using *ffmpeg* did the trick for me:

`adb forward tcp:8080 tcp:8080 && ffmpeg -i http://127.0.0.1:16754/video -vf "transpose=1" -pix_fmt yuv420p -f v4l2 /dev/video2`

(`transpose=1` is for rotation, you might want to use another value; also, as stated above by others, make sure to `killall adb` in case an instance from a previous run of the script is still running)

sorachan, 2020-04-17:
I ended up circumventing the problem by not using the script / GStreamer. Using *ffmpeg* did the trick for me:

`adb forward tcp:8080 tcp:8080 && ffmpeg -i http://127.0.0.1:8080/video -vf "transpose=1" -pix_fmt yuv420p -f v4l2 /dev/video2`

(`transpose=1` is for rotation, you might want to use another value; also, as stated above by others, make sure to `killall adb` in case an instance from a previous run of the script is still running)

basiliscus79, 2020-04-18:
I tried, but didn't work.

Thanks

gpatel-fr, 2020-04-19:
Thanks @sorachan I was already playing with ffmpeg but missed the idea of starting adb manually. Building on your idea, I can now use

killall adb ; sleep 2 ; adb forward tcp:8080 tcp:8080 && ./ffmpeg -flags +low_delay  -rtsp_transport tcp  -i rtsp://127.0.0.1:8080/video/h264 -f v4l2 /dev/video0

replacing IPWebcam with Camon; I get the impression that with my antique phone the performance, quality and stability is better using Camon, to the point that this old thing can actually be used (my previous attempt could only be used for testing my video server). The only problem is a slight delay of about 1/2 second, but it's acceptable until I can finally get out to buy a new webcam.

oscarsantis, 2020-05-11:
Hi, thanks for your work.
I can run ip-camera on Debian 10 with:
- set android device IP on prepare-videochat.sh
- run sudo modprobe v4l2loopback
- run ./prepare-videochat.sh
after to answer two pop-ups camera works fine.
My question is:
How can I simplify the script after install all libraries after run server for the first time?

bluezio, 2020-06-07:
The script should not try to install anything if it detects all the dependencies are there. If you find it is redoing an installation, then it probably means that some dependency is not where the script expected it. Pull requests for that would be welcome :-).

closed 2020-06-07 by bluezio