agarciadom/ipwebcam-gst #64: Script ended without error or warning and not able to make it work

Phar71, 2020-04-02:
Hello,

First, thank you for your work.
Just trying to use the script but seems to do just nothing in my case.

**OS:** Arch Linux
**uname -a:** Linux hideqel 5.5.13-arch2-1 #1 SMP PREEMPT Mon, 30 Mar 2020 20:42:41 +0000 x86_64 GNU/Linux
**trizen -Si v4l2loopback-dkms**
Repository      : AUR
Name            : v4l2loopback-dkms
Version         : 0.12.3-2

**IP webcam** installed on an Android phone connected by wifi at 192.168.0.13:8080 --> ok in a webpage
**Followed the troubleshooting section of the script**

1. v4l2loopback returns a window with colors from green to multicolor
2. video connection is ok, I'm able to see me on the new window (awfull :-)
3. is not relevant for me as I use the phone through wifi

when I execute the script, I have no message on the terminal and that just ended.
I tried to look in Teams, Zoom and I am able to find the /dev/video0 created but it returns just a black screen.

Tell me if you need some other informations.
Thank for your reading and help
Philip

davemulford, 2020-04-08:
I'm having similar issues, except when I run the script, it doesn't exit; I get a message that says, *"Press enter to end stream"*.

Like you, items 1 and 2 from the TROUBLESHOOTING section in the script work great. I'm not connecting multiple android devices to my PC, so item 3 also doesn't apply to me. However, my issue seems to be with the v4l2sink gstreamer element not working.

I found this out by looking in the `feed.log` file that gets generated when the script calls `gst-launch-1.0`.  If you have that file, does it look like the following?

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'souphttpsrc0': gst.soup.session=context, session=(SoupSession)NULL, force=(boolean)false;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
WARNING: from element /GstPipeline:pipeline0/GstDecodeBin:decodebin0: Delayed linking failed.
Additional debug info:
./grammar.y(506): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstDecodeBin:decodebin0:
failed delayed linking some pad of GstDecodeBin named decodebin0 to some pad of GstVideoConvert named videoconvert0
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstDecodePad:src_0.GstProxyPad:proxypad1: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
streaming stopped, reason not-linked (-1)
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
```

Adding `--gst-debug="v4l2sink:5"` to the `gst-launch-1.0` command doesn't output anything that makes the issue obvious to me, but I'm not a gstreamer dev!

My plan is to find some forum where I can get help from a gstreamer expert, who can perhaps read a debug log file and tell me what I'm doing wrong.

davemulford, 2020-04-09:
With help from a friend, I was able to figure out my issue &ndash; permissions on the `/dev/video0` device.

Here are the permissions for my video device.

    $ ls -l /dev/video0
    crw-rw----+ 1 root video 81, 0 Apr  9 14:39 /dev/video0

I added my user to the `video` group with the following command.

    $ sudo usermod -a -G video <username>

This is what allowed me to get past the *"GstSoupHTTPSrc:souphttpsrc0: Internal data stream error"* message, which is actually telling me about a permissions issue.

The full `gst-launch-1.0` on my system looks like:

    $ gst-launch-1.0 -v souphttpsrc location=http://127.0.0.1:8080/videofeed do-timestamp=true is-live=true ! queue ! multipartdemux ! decodebin ! videoconvert ! v4l2sink device=/dev/video0 sync=true

**Note:** I added the `queue` element to place the `souphttpsrc` and the rest of the pipeline on separate threads for performance reasons.  I also removed the `videorate` element as it was causing the pipeline to constantly pause and unpause, resulting in a very delayed video stream.

With the above command, I'm able to see a near zero lag stream.

I also added the `exclusive_caps=1` option when loading the `v4l2loopback` kernel module to allow the video device to be seen by chrome and other apps. This is suggested in [v4l2loopback issue #78](https://github.com/umlaeute/v4l2loopback/issues/78).

Launching the IPWebcam app on my phone is also failing with the script, and using the following seems to work for me.

    $ adb shell monkey -p com.pas.webcam -c android.intent.category.LAUNCHER 1

I hope this is useful for you, and doesn't just add more noise to the conversation. I'm thinking about rewriting the script into a python app that works with GStreamer libraries, instead of relying on the `gst-launch-1.0` command.

bluezio, 2020-04-14:
Thank you for looking into this, @davemulford. I don't have as much time as before to debug this script on all the various configurations people have :-).

I have made some improvements to the script based on your tweaks: dropped `videorate`, added `queue`, and added the `exclusive_caps=1` option to the `modprobe` call. I have also added read/write permission checks right after the device autodetection, which should hopefully make it easier for people to find out if they missed that part of their setup. GStreamer error messages can be very cryptic :-/.

bluezio, 2020-04-14:
Closing since @davemulford got the script to work after some changes, most of which have been incorporated into the script. @Phar71, feel free to reopen if you still cannot get this to work on the latest version of the script.

closed 2020-04-14 by bluezio