agarciadom/ipwebcam-gst #68: Script doesn't work with latest IP Webcam release due to security exception

sorachan, 2020-04-17:
Using the latest version of IP Webcam from Google Play Store, the script didn't work for me because the `android:exported` property of `com.pas.webcam.Rolling` is set to `false`, leading the script to terminate with the following message:

```
Starting: Intent { act=android.intent.action.MAIN cmp=com.pas.webcam/.Rolling }
Security exception: Permission Denial: starting Intent { act=android.intent.action.MAIN flg=0x10000000 cmp=com.pas.webcam/.Rolling } from null (pid=16880, uid=2000) not exported from uid 10208

java.lang.SecurityException: Permission Denial: starting Intent { act=android.intent.action.MAIN flg=0x10000000 cmp=com.pas.webcam/.Rolling } from null (pid=16880, uid=2000) not exported from uid 10208
        at com.android.server.am.ActivityStackSupervisor.checkStartAnyActivityPermission(ActivityStackSupervisor.java:2028)
        at com.android.server.am.ActivityStarter.startActivity(ActivityStarter.java:923)
        at com.android.server.am.ActivityStarter.startActivity(ActivityStarter.java:656)
        at com.android.server.am.HwActivityStarter.startActivity(HwActivityStarter.java:222)
        at com.android.server.am.ActivityStarter.startActivityMayWait(ActivityStarter.java:1438)
        at com.android.server.am.HwActivityStarter.startActivityMayWait(HwActivityStarter.java:412)
        at com.android.server.am.ActivityStarter.execute(ActivityStarter.java:597)
        at com.android.server.am.ActivityManagerService.startActivityAsUser(ActivityManagerService.java:5759)
        at com.android.server.am.ActivityManagerService.startActivityAsUser(ActivityManagerService.java:5733)
        at com.android.server.am.ActivityManagerShellCommand.runStartActivity(ActivityManagerShellCommand.java:479)
        at com.android.server.am.ActivityManagerShellCommand.onCommand(ActivityManagerShellCommand.java:161)
        at android.os.ShellCommand.exec(ShellCommand.java:103)
        at com.android.server.am.ActivityManagerService.onShellCommand(ActivityManagerService.java:17744)
        at android.os.Binder.shellCommand(Binder.java:642)
        at android.os.Binder.onTransact(Binder.java:540)
        at android.app.IActivityManager$Stub.onTransact(IActivityManager.java:3570)
        at com.android.server.am.ActivityManagerService.onTransact(ActivityManagerService.java:3654)
        at com.android.server.am.HwActivityManagerService.onTransact(HwActivityManagerService.java:609)
        at android.os.Binder.execTransact(Binder.java:739)

```
What did the trick for me was downloading IP Camera APK from APKpure (version v1.14.36.755, same as Play Store) and using *apktool* to unpack the APK, changing the property value inside *androidManifest.xml* to `true` and repackaging the APK. I know this isn't your app, I'm forwarding the issue to the developer, I'm just leaving this here to let you know the script is currently broken.

(The operating system used was Manjaro Linux but this should be irrelevant, the Android version is Android 9 / EMUI 9.1.0.260 running on a Huawei P30 Pro.)

bluezio, 2020-04-18:
Thank you for reporting this issue! I'll leave it open here in case other people have the same problem.

craigteegarden, 2020-04-23:
This issue also happened for me, but if you manually start the app on the phone first the script will continue past this point. Kind of ruins a lot of the push-button automation though.

bluezio, 2020-04-24:
After talking with the IP Webcam, the script has now been modified so that it asks explicitly the user to start the camera on the app.

closed 2020-04-24 by bluezio