agarciadom/ipwebcam-gst #69: Pi 4 script works but no device recognized

basiliscus79, 2020-04-18:
Hello, I am using VNC Viewer from my Macbook and access the desktop of a Pi4 in the same LAN. 
VNC does not support audio so I decided to try streaming audio to an external device such as a Smartphone (Galaxy S4) that I could use as a wireless webcam at the same time.
I installed the IPWebcam app on the phone and tried to change the IP address in the prepare-videochat.sh script in order to match it to the IP Webcam address.
With a lot of surprise, the scripts itself seems to work, i.e. after running the IPWebcam server and then the script, I get to the message saying the audo and video are streaming.
I could not install Android SDK and I get the warning, but it seems to work anyways.

IP Webcam is running fine in Chromium and I am able to record both video and audio from the web.
Running pavucontrol show all new the devices but they are actually not recognized by any application in the OS and even if I try to use google Meet in Chromium it does not see any device.
Any suggestion?

Thanks so much
Michele

bluezio, 2020-04-18:
Please check this comment: https://github.com/bluezio/ipwebcam-gst/issues/63#issuecomment-608455047

basiliscus79, 2020-04-18:
I tried, but did not work.

THX

bluezio, 2020-04-19:
What did you get? Can't do much without more information :-).

basiliscus79, 2020-04-19:
Actually I did not get much, it just worked as it did before but at the end of the day I still cannot detect any device. I am pretty sure that I am doing something wrong, but I cannot understand what.
As soon as I have the time I will test whether I get the same problems if I use the Pi directly (not from remote connection), so that I can find if the issue relates much to the remote connection or to the software/OS itself.
I will keep you up to date, many thanks for your help.

Best
Michele

bluezio, 2020-04-28:
Closing due to inactivity - feel free to reopen once you have more information.

closed 2020-04-28 by bluezio