agarciadom/ipwebcam-gst #71: Space runs out on ./prepare-videochat.sh

hatlesshacker, 2020-04-25:
Everything seems fine, the script even shows that the webcam is ready to be used, but soon after, the space on the system runs out and the system finally crashes.

Debian 32 bit.

bluezio, 2020-04-26:
The script should only be writing to your /dev/video* device. Have you tried using a disk space analysis tool to see where the disk space is being used?

hatlesshacker, 2020-04-27:
The system hangs within a few seconds of running the script.. Within that time I somehow managed to see the amount of free space in nautilus FM. I don't think there's any other way for me to know where the space gets used up.

Also, on reboot, the system comes back to normal. Say I have 2GB free space before running the script. When the system reboots after crashing, It still has 2GBs.

bluezio, 2020-04-28:
It sounds like something is being written to `/tmp`, which gets cleared after each reboot (explaining why you get the disk space back upon rebooting). (It really shouldn't be written - the script doesn't do anything like it by itself. Might be a GStreamer issue?)

I'd suggest running something like this in another terminal before you run the script, so we can get more info:

```
rm -f $HOME/tmp_usage
while true
do
    du -c /tmp >> $HOME/tmp_usage
    sleep 0.1s
done
```

This way, upon reboot, you should be able to look at `$HOME/tmp_usage` and see where the disk space was piling up before the crash and your reboot.

s0menickname, 2020-05-05:
Same issue, but reboot did not fix it. Arch Linux, amd64.

bluezio, 2020-06-07:
Unfortunately, I cannot reproduce the issue in my system (Ubuntu 19.10, kernel 5.3.0-55-generic, v4l2loopback-dkms 0.12.2-1), and I do not have enough information to trace where it may be coming from. Please reopen if you find out more information.

closed 2020-06-07 by bluezio

hatlesshacker, 2020-06-08:
I ran the shell script, but with no result. I believe this problem is due to something on my system specifically. Thanks for the replies :-)

andreasvc, 2021-01-24:
I ran into the same problem. I believe the problem is caused by this line of `run-videochat.sh`:
```
aec_method="webrtc" save_aec=true use_volume_sharing=true) || true
```

save_aec=true causes files to be written to /tmp.
See https://cgit.freedesktop.org/pulseaudio/pulseaudio/tree/src/modules/echo-cancel/module-echo-cancel.c#n73

It seems the fix is to disable `save_aec`.