agarciadom/ipwebcam-gst #73: Could not install the v4l2loopback kernel module through apt-get. ubuntu 18.04

nmolanog, 2020-04-28:
doing `./prepare-videochat.sh ` gets the following output:

`
find: ‘/lib/modules/5.3.0-46-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-46-generic/extramodules’: No such file or directory
Trying to install v4l2loopback-dkms package.
[sudo] password for nicolas: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
v4l2loopback-dkms is already the newest version (0.10.0-1ubuntu1.2).
The following packages were automatically installed and are no longer required:
  efibootmgr gir1.2-appindicator3-0.1 gir1.2-geocodeglib-1.0 libfwup1 libllvm8
  libllvm8:i386 redshift ubuntu-web-launchers
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
find: ‘/lib/modules/5.3.0-46-generic/extra’: No such file or directory
find: ‘/lib/modules/5.3.0-46-generic/extramodules’: No such file or directory
`
also a windows pops up with the following message: 
`Could not install the v4l2loopback kernel module through apt-get.`

output for dpkg -l | grep v4l2loopback-dkms:

`ii v4l2loopback-dkms 0.10.0-1ubuntu1.2 all Source for the v4l2loopback driver (DKMS)`

fal-pasa, 2020-04-30:
Your problem should be the same as shown in the #63

nmolanog, 2020-05-02:
> Your problem should be the same as shown in the #63

I published there, and was told to open a new issue

bluezio, 2020-05-02:
Yes, because that issue is closed and the messages are slightly different :-). I'm not so sure it will be exactly the same issue.

jeanmarie18, 2020-05-03:
Hi,

My OS is Ubuntu 18.04.4.

Before testing "IP Webcam" I installed DroidCam: I could use the camera in Zoom but not the microphone and I need it => so I first uninstalled DroidCam because it had its own V4L2 driver named "v4l2droidcam-dc" (I used the developer uninstall script which did the job)

After that I launched your script prepare-videochat.sh: here is the output:

```
find: ‘/lib/modules/4.15.0-99-generic/extra’: No such file or directory
find: ‘/lib/modules/4.15.0-99-generic/extramodules’: No such file or directory
Trying to install v4l2loopback-dkms package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libllvm7 libllvm8
Use 'sudo apt autoremove' to remove them.
Suggested packages:
  v4l2loopback-utils
The following NEW packages will be installed:
  v4l2loopback-dkms
0 upgraded, 1 newly installed, 0 to remove and 12 not upgraded.
Need to get 23,3 kB of archives.
After this operation, 112 kB of additional disk space will be used.
Get:1 http://fr.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 v4l2loopback-dkms all 0.10.0-1ubuntu1.2 [23,3 kB]
Fetched 23,3 kB in 0s (183 kB/s)              
Selecting previously unselected package v4l2loopback-dkms.
(Reading database ... 203048 files and directories currently installed.)
Preparing to unpack .../v4l2loopback-dkms_0.10.0-1ubuntu1.2_all.deb ...
Unpacking v4l2loopback-dkms (0.10.0-1ubuntu1.2) ...
Setting up v4l2loopback-dkms (0.10.0-1ubuntu1.2) ...
Loading new v4l2loopback-0.10.0 DKMS files...
Building for 4.15.0-99-genericsudo apt install v4l-utils
Building initial module for 4.15.0-99-generic
Done.

v4l2loopback:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.15.0-99-generic/updates/dkms/

depmod.......

DKMS: install completed.
find: ‘/lib/modules/4.15.0-99-generic/extra’: No such file or directory
find: ‘/lib/modules/4.15.0-99-generic/extramodules’: No such file or directory
````
And simultaneously I had a window with the title "Error" and with the message "**Could not install the v4l2loopback kernel module through apt-get**."

I did `dpkg -l | grep v4l2loopback-dkms` which displayed:
```
ii  v4l2loopback-dkms                          0.10.0-1ubuntu1.2                                all          Source for the v4l2loopback driver (DKMS)
```
Then I typed: `lsmod|grep v4l` which gave an empty result, so I did:
`sudo insmod v4l2loopback`
Then `lsmod|grep v4l` gives a correct answer:
```
v4l2loopback           40960  0
videodev              184320  1 v4l2loopback
```
`ls -l /dev/video*` seems correct:
```
crw-rw----+ 1 root video 81, 0 mai    2 14:44 /dev/video0
```
I installed pavucontrol but in the "Input devices" there is no this "null sink".

I launched the server on "IP Webcam"

I then launched this gstream command:
```
gst-launch-1.0 souphttpsrc location="http://192.168.1.11:8080" do-timestamp=true is-live=true ! multipartdemux ! jpegdec ! ffmpegcolorspace ! ximagesink
```
but it only displays `"WARNING: erroneous pipeline: no element "ffmpegcolorspace"`

launched pavucontrol and in the "Input devices" there is still no this "null sink".
so I removed the ffmpegcolorspace option and therfore typed:
```
gst-launch-1.0 souphttpsrc location="http://192.168.1.11:8080" do-timestamp=true is-live=true ! multipartdemux ! jpegdec ! ximagesink
```
then the messages are:
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'souphttpsrc0': gst.soup.session=context, session=(SoupSession)NULL, force=(boolean)false;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstMultipartDemux:multipartdemux0: Could not demultiplex stream.
Additional debug info:
multipartdemux.c(506): multipart_parse_header (): /GstPipeline:pipeline0/GstMultipartDemux:multipartdemux0:
Boundary not found in the multipart header
Execution ended after 0:00:00.598077940
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```
looking at pavucontrol and still no "null sink"

I searched on the net the "ERROR: from element /GstPipeline:pipeline0/GstMultipartDemux:multipartdemux0: Could not demultiplex stream." and is it possible that only one of the two stream (audio or video) reaches the computer ?

What can I do then ?
(I will try to learn more about there GSstreamer pipelines as don't know what it is.)

nicola-lunghi, 2020-05-20:
Hi
the issue is that the module on ubuntu 20.04 and 18.04 is called v4l2loopback.ko not
so the grep `find /lib/modules/$(uname -r)/ -name "$MODULE.ko.*" doesn't find it

``
find /lib/modules/5.4.0-29-lowlatency/ -name 'v4l2loopback.ko'
/lib/modules/5.4.0-29-lowlatency/updates/dkms/v4l2loopback.ko
``

can be fixed replacing the grep with

``find /lib/modules/$(uname -r)/ -name "$MODULE.ko*" ``

pablorq, 2020-05-22:
After the error I did this:

`sudo modprobe v4l2loopback`

And then it seems to work, but I don't have any video nor audio. Still working on that...

The workaround is from [this issue](https://github.com/bluezio/ipwebcam-gst/issues/63#issuecomment-626937255)

jeanmarie18, 2020-05-30:
> Hi
> the issue is that the module on ubuntu 20.04 and 18.04 is called v4l2loopback.ko not
> so the grep `find /lib/modules/$(uname -r)/ -name "$MODULE.ko.*" doesn't find it
>
> `find [...]

Hello and thank you for your help: so I removed the dot before the star, put the correct Wifi IP address in the script and relaunched it (sudo ./prepare-videochat.sh). Here is the output:
```
Trying to install gstreamer1.0-pulseaudio package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-pulseaudio is already the newest version (1.14.5-0ubuntu1~18.04.1).
The following packages were automatically installed and are no longer required:
  libllvm7 libllvm8
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 19 not upgraded.
Home directory not accessible: Permission denied
No PulseAudio daemon running, or not running as session daemon.
Home directory not accessible: Permission denied
No PulseAudio daemon running, or not running as session daemon.
Home directory not accessible: Permission denied
No PulseAudio daemon running, or not running as session daemon.
Home directory not accessible: Permission denied
No PulseAudio daemon running, or not running as session daemon.
Failure: Module initialization failed
```
Then I have the information "IP Webcam audio is streaming through pulseaudio sink 'ipwebcam'.
IP Webcam video is streaming through v4l2loopback device /dev/video0.
 You can now open your videochat app." => I pressed OK.
Then the terminal continues with:
```
Press enter to end stream
```
I can now see in "pavucontrol":
- Output Devices / IP Webcam
- Input Devices / Monitor of IP Webcam

I then launched "guvcview": the video works :-)
However no sound is output to the speakers...
Any idea ?

I will try later with zoom.

bluezio, 2020-06-07:
The `.ko.*` issue is now fixed: you shouldn't have to manually tweak the script anymore.

For the sound, you won't hear it through your speakers. The sound works like a new microphone, so you can tell Zoom to use your phone as the microphone in the video call.

closed 2020-06-08 by bluezio