agarciadom/ipwebcam-gst #74: No Video being displayed

LuD1161, 2020-05-01:
Hi @bluezio ,

Thanks for the script.
I am not able to get any video output, it just shows the first frame captured by the camera and then freezes.
However the audio looks fine.

Please find the below attached images :
[image]

The device displayed as 'Dummy video device'
[image]

bluezio, 2020-05-01:
There are some issues with the version of v4l2loopback packaged in some recent distributions. Some users have reported success by using a newer version of the module. Could you check this comment and let me know if it works for you?

https://github.com/bluezio/ipwebcam-gst/issues/63#issuecomment-608455047

LuD1161, 2020-05-01:
Hi @bluezio 

I was trying out droidcam and it handles video input easily. It also has a script which installs and uses v4l2loopback but there's no pulseaudio ( which is still an open issue ) and hence I started with this.
Maybe looking at their script might give you some idea as to what different they're doing to make the video work ( cause I have no idea about these and I did try fiddling around whole night yesterday ). 

I tried changing the device to the device of droidcam i.e. `/dev/video3` and using it's v4l2loopback but in vain. I don't know understand much, so I was just fiddling around with the parameters to make it work :sweat_smile: 

I added it's module path in your shell code so that your code could find it's driver : ( in the `has_kernel_module` function )
```shell
find /lib/modules/$(uname -r)/kernel/drivers/media/video/ -name "$MODULE.ko.*" | egrep '.*'||
```
However that didn't work out.

```shell
#!/bin/bash
#
# Usage: ./install [640 480]
# Dev47Apps.com
#
set -e
V4L2_LOOPBACK_DIR="v4l2loopback";
V4L2_LOOPBACK_DC="v4l2loopback_dc";
V4L2_LOOPBACK_KO="v4l2loopback-dc.ko";
DRIVER_LOCATION="/lib/modules/`uname -r`/kernel/drivers/media/video/";

WIDTH="640"
HEIGHT="480"

if [[ "$1" != "" && "$2" != "" ]]
then
	WIDTH=$1
	HEIGHT=$2
fi

if [[ ! $(echo "$WIDTH" | grep -E "^[0-9]+$") || ! $(echo "$HEIGHT" | grep -E "^[0-9]+$") ]]
then
	echo "Inavlid parameters: '$WIDTH' and '$HEIGHT'"
	exit 1
fi

if [[ $WIDTH -lt 240 || $HEIGHT -lt 160 ]]
then
	echo "Parameters too low: '$WIDTH' and '$HEIGHT'"
	exit 1
fi

echo "Webcam parameters: '$WIDTH' and '$HEIGHT'"

echo "Building $V4L2_LOOPBACK_KO"
make -C $V4L2_LOOPBACK_DIR

if [ ! -e "$V4L2_LOOPBACK_DIR/$V4L2_LOOPBACK_KO" ]
then
	echo "$V4L2_LOOPBACK_KO not built.. Failure"
	exit 1;
fi

echo "Moving driver and executable to system folders"
if [ ! -e $DRIVER_LOCATION ]
then
	mkdir -p $DRIVER_LOCATION
fi

cp "$V4L2_LOOPBACK_DIR/$V4L2_LOOPBACK_KO" $DRIVER_LOCATION
cp droidcam /usr/bin/
cp droidcam-cli /usr/bin/

echo "Registering webcam device"
modprobe videodev
insmod $DRIVER_LOCATION$V4L2_LOOPBACK_KO width=$WIDTH height=$HEIGHT

echo "Running depmod"
depmod -a
make -C $V4L2_LOOPBACK_DIR clean

echo "Adding uninstall script"
mkdir -p /opt || true
cp uninstall /opt/droidcam-uninstall

etc_modules() {
	echo "Adding driver to /etc/modules"
	cp /etc/modules /etc/modules.bak
	prevperm=`stat -c %a /etc/modules`
	chmod 666 /etc/modules
	[[ ! $(egrep "^videodev$" /etc/modules) ]] && echo "videodev" >> /etc/modules
	[[ ! $(egrep "^$V4L2_LOOPBACK_DC" /etc/modules) ]] && echo "$V4L2_LOOPBACK_DC" >> /etc/modules
	chmod $prevperm /etc/modules
}

etc_modules_load_d() {
	printf "videodev\n$V4L2_LOOPBACK_DC\n" \
		 > /etc/modules-load.d/droidcam.conf

}

[[ -d "/etc/modprobe.d/" ]] && echo "options $V4L2_LOOPBACK_DC width=$WIDTH height=$HEIGHT" > /etc/modprobe.d/droidcam.conf

if [ -e "/etc/modules" ]
then
	etc_modules
elif [ -d "/etc/modules-load.d" ]
then
	etc_modules_load_d
else
	echo "Warning: Unknown distro. Webcam module may not load after a reboot :("
fi

echo "Done"
```

LuD1161, 2020-05-01:
Hi @bluezio 
I did try, but that doesn't seem to work.

LuD1161, 2020-05-01:
Also, is there any way to just transfer **only** the audio using USB without opening the app.
Because currently I've stick together a setup in which ip-webcam sends only the audio and droidcam is used for transmitting the video stream.

However, the problem with this setup is it's heavy on the phone and moreover the sound is very low.

Thanks :slightly_smiling_face:

LuD1161, 2020-05-01:
So, finally I got this running with video input from droidcam and audio using [mic_over_mumble](https://github.com/pzmarzly/mic_over_mumble).

bluezio, 2020-06-07:
Thanks for letting us know!

closed 2020-06-07 by bluezio