agarciadom/ipwebcam-gst #80: Cannot get feed from v4l2loopback

Harvara, 2020-05-25:
I worked my way through the setup and had a few problems due secure boot and needed to sign the modules by myself. Now I can run the prepare file (I still get PulseAudio errors) and get the message my device is ready at /dev/video04 and "Press enter to end stream". 

If I stop it, is get:

    You have to specify a module index or name

But if I try to use it, I just get a black feed in Chrome/Teams and an error in vlc. I tried another setup where i could successfully stream my desktop as webcam signal, but it wont work with the ipwebcam-server.

My feed.log has:

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'souphttpsrc0': gst.soup.session=context, session=(SoupSession)NULL, force=(boolean)false;
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:sink: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstQueue:queue0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)1
Redistribute latency...
/GstPipeline:pipeline0/GstPulseSink:pulsesink0.GstPad:sink: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)1
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
WARNING: from element /GstPipeline:pipeline0/GstDecodeBin:decodebin0: Delayed linking failed.
Additional debug info:
./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstDecodeBin:decodebin0:
failed delayed linking some pad of GstDecodeBin named decodebin0 to some pad of GstVideoConvert named videoconvert0
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstDecodePad:src_0.GstProxyPad:proxypad1: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc1: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc1:
streaming stopped, reason not-linked (-1)
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
ERROR: from element /GstPipeline:pipeline0/GstQueue:queue1: Internal data stream error.
Additional debug info:
gstqueue.c(988): gst_queue_handle_sink_event (): /GstPipeline:pipeline0/GstQueue:queue1:
streaming stopped, reason not-linked (-1)
```

bluezio, 2020-06-07:
Which OS are you using? I've never seen this issue before.

Harvara, 2020-06-08:
Ubuntu 18.04.4 LTS

k-five, 2020-07-27:
I am using this awesome script for just two days and it works perfectly. Also the OS I am using is `Ubuntu 18' so you should not have any problem . First run script for installing some dependencies and the after the first time, second you can run it the simulate a virtual video device. I am using with Skype and it works fine.

output for the second time

```
Trying to install gstreamer1.0-pulseaudio package.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-pulseaudio is already the newest version (1.14.5-0ubuntu1~18.04.1).
The following packages were automatically installed and are no longer required:
  cmake-data libjsoncpp1 librhash0 libuv1
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 107 not upgraded.
Please start IP Webcam or IP Webcam Pro server and press Enter

Press enter to end stream
```

bluezio, 2020-08-14:
Closing due to inactivity from @Harvara.

closed 2020-08-14 by bluezio