agarciadom/ipwebcam-gst #82: Ported script to perl

Duoxilian, 2020-06-01:
I needed some configuration options for my setup, and I found it easier to understand the script by porting it to perl. If you are interested, I can post it. I have only tested it on ubuntu 18.04, though.

bluezio, 2020-06-07:
I can have a look :-). I'd only be able to merge it if it was a one-to-one replacement of the current script (including the Arch-specific bits), however. If it's substantially different, it may be better to create a new project for it.

Duoxilian, 2020-06-15:
Sorry for the delay. I ported your "arch-specific bits" but without testing. However, there are extensions, as I wanted to control more of the webcam from the parameters of the script rather than having to make changes in the script itself.

bluezio, 2020-08-14:
Closing this, as @battarsa kindly split the script into two and it's a bit more manageable now, still in Bash :-). @battarsa also contributed command-line flags for the "run" half.

closed 2020-08-14 by bluezio