agarciadom/ipwebcam-gst #84: Camera is missing

katsar0v, 2020-06-07:
I got this message:

```
IP Webcam audio is streaming through pulseaudio sink 'ipwebcam'.
IP Webcam video is streaming through v4l2loopback device /dev/video0.
 You can now open your videochat app.
```

But there is no camera available in skype to choose from. I have previously installed droidcam (and maybe not properly uninstalled) so there might be a collision. Can somebody help me?

bluezio, 2020-06-07:
After the script says the IP Webcam video/audio is streaming, try listing your video devices:

```sh
ls /dev/video*
```

Then try to find out if the v4l2loopback devices has been set up:

```sh
sudo v4l2-ctl -D -d /dev/videoX | grep loop
```

For instance, in my case, I get this from my `/dev/video4`:

```
	Driver name      : v4l2 loopback
	Bus info         : platform:v4l2loopback-000
```

Normally the script should be smart enough to pick the `/dev/videoX` that uses v4l2loopback. If that is not working, try uncommenting the line with `#DEVICE=/dev/video0` in the script and set it to the right value.

katsar0v, 2020-06-07:
v4l2 modbro had to be reloaded, solved. One side question: is v4l2 limited to 720p? Once I go to 1080p, discord and skype show only black screen

bluezio, 2020-06-08:
The script has WIDTH and HEIGHT variables for the width and height of the video. Did you try tweaking those?

katsar0v, 2020-06-08:
I did, I know. It works with 720p, but on 1080p the camera is gone (in skype and discord)

apandada1, 2020-08-10:
@katsar0v I have the same issue. How did you reload v4l2 modprobe ?

bluezio, 2020-08-14:
The `install-videochat.sh` script tells you how to do that (comments at the beginning of the script). Since you do not want multiple devices, it'd be just this:

```sh
sudo modprobe -r v4l2loopback
sudo modprobe v4l2loopback exclusive_caps=1
```

bluezio, 2021-04-12:
Closing due to inactivity.

closed 2021-04-12 by bluezio