agarciadom/ipwebcam-gst #86: Support for Raspberry Pi (raspbian)

Longoon12000, 2020-06-20:
Hello

I can't get this tool to run on Raspbian GNU/Linux 10 (buster) armv7l.

First, I rewrote the usage of zenity as I run the raspberry pi headless.

After initially running the tool, successive runs try to do `install_package "gstreamer${GST_VER}-pulseaudio"` because libgstpulseaudio.so and libgstpulse.so weren't found at the search paths. They're located at `/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/` but `uname -m` outputs armv71. I've symlinked `/usr/lib/arm-linux-gnueabihf/gstreamer-1.0/` to `/usr/lib/gstreamer-1.0/` to work around this issue.

Next pulseaudio itself and pulseaudio-utils, which contains pacmd and pactl, is not installed by default and not checked by the script. Manually installing it and starting `pulseaudio -D` fixes
```
./prepare-videochat.sh: line 522: pacmd: command not found
./prepare-videochat.sh: line 523: pacmd: command not found
./prepare-videochat.sh: line 318: pacmd: command not found
./prepare-videochat.sh: line 318: pacmd: command not found
./prepare-videochat.sh: line 534: pactl: command not found
```
and
```
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
No PulseAudio daemon running, or not running as session daemon.
Connection failure: Connection refused
pa_context_connect() failed: Connection refused
```

Now I am getting `IP Webcam audio is streaming through pulseaudio sink 'ipwebcam'.\nIP Webcam video is streaming through v4l2loopback device /dev/video0.\n You can now open your videochat app.` however the device /dev/video0 doesn't seem to be setup properly:

```
$ ffmpeg -i /dev/video0
[video4linux2,v4l2 @ 0x1a6a170] Not a video capture device.
/dev/video0: No such device
$ file /dev/video0
/dev/video0: character special (81/0)
```

I've found a few similar issues online, they suggest `# insmod v4l2loopback exclusive_caps=1` which doesn't work:
```
$ sudo insmod v4l2loopback exclusive_caps=1
insmod: ERROR: could not load module v4l2loopback: No such file or directory
$ lsmod | grep v4l2loopback
v4l2loopback           32768  0
v4l2_common            16384  2 bcm2835_v4l2,v4l2loopback
videodev              200704  7 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2,v4l2loopback
```

I am unsure how to proceed from here. Any suggestions?

adremides, 2020-06-30:
Did you installed the kernel module?
Maybe [here (Raspberry pi fail to install #84)](https://github.com/umlaeute/v4l2loopback/issues/84) you can find some answer.

bluezio, 2020-08-14:
Closing due to inactivity. Please reopen if you want to work a bit more on this :-).

closed 2020-08-14 by bluezio