agarciadom/ipwebcam-gst #91: No new video device shows up after running ./run-videochat.sh

apandada1, 2020-08-14:
OS: Linux Mint 19.3

I am using the [latest code](https://github.com/bluezio/ipwebcam-gst/commit/e993f58a0950a8c69a98d3f2e70d1978446a4f34).

```
$ sudo ./install-videochat.sh 
videochat installed, launch with './run-videochat.sh'
```
```
$./run-videochat.sh 
```
 Phone is connected to computer via USB, and I have USB debugging enabled, `adb` installed in the computer, and IP Webcam server is turned on in phone.
This dialog appeared.
```
IP Webcam audio is streaming through pulseaudio sink 'ipwebcam'.
IP Webcam video is streaming through v4l2loopback device /dev/video2.
You can now open your videochat app.
```

Press enter to end stream
```

I opened Cheese, only my internal webcam shows up.

After pressing enter I get
```
./run-videochat.sh: line 455:  7003 Terminated              "$GSTLAUNCH" -e -vt --gst-plugin-spew --gst-debug="$GST_DEBUG" $PIPELINE > feed.log 2>&1
Disconnected from IP Webcam. Have a nice day!
```
Am I missing something simple?

jonasmalacofilho, 2020-08-20:
You probably need to enable `exclusive_caps=1` for Cheese (and Chrome) to work correctly with it.

```
$ # manually unload the v4l2loopback module
$ # (only needed if already loaded without the necessary parameter)
$ sudo modprobe -r v4l2loopback

$ # use V4L2_OPTS to pass extra parameters when loading v4l2loopback
$ V4L2_OPTS=exclusive_caps=1 ./run-videochat.sh 
```

For more information, see umlaeute/v4l2loopback/#78.

apandada1, 2020-08-26:
I still get the same error
```
V4L2_OPTS=exclusive_caps=1 ./run-videochat.sh 
Loading module
modprobe: ERROR: could not insert 'v4l2loopback': Bad address
```

mail4me9999, 2020-10-09:
this error is due to incompatibility between the v4l2loopback and the kernel.
need to recompile the module from source. this is not a problem related to this script

apandada1, 2021-01-11:
I am closing this as the solution in #100 worked for me. Please add the solution proposed in #100 to the README. Also, I added this line (https://github.com/apandada1/ipwebcam-gst/blob/master/run-videochat.sh#L278) for my own use. I don't know if that may break functionality outside Ubuntu 20.04. Can it be selectively loaded with `exclusive_caps=1` in Ubuntu?

closed 2021-01-11 by apandada1