agarciadom/ipwebcam-gst #92: help please

esquee, 2020-08-19:
when I run the command:
sudo: ./install-videochat.sh: command not found

what do I do?
thanks!

bluezio, 2020-08-19:
The right command is `sudo ./install-videochat.sh`, with no : in the middle. Try it out :-).

esquee, 2020-08-20:
> The right command is `sudo ./install-videochat.sh`, with no : in the middle. Try it out :-).
Hello! 
That's the command I run but I still get the same response- command not found. I attached the screenshot. 

[image]

rafasantos, 2020-08-27:
@esquee are you sure you issued the command from the folder where you downloaded the `install-video-chat.sh`. It looks like the file is simply missing from the folder (currently at `~` which is your home folder).

bluezio, 2021-04-12:
Closing due to inactivity.