agarciadom/ipwebcam-gst #97: get only single frame

mail4me9999, 2020-10-09:
When I use the current version of run-videochat.sh (today is 2020-10-09)

on ubuntu 18.04 with v4l2loopback driver version 0.12.5  compiled manually
and gstreamer-1.0
I get only the first frame from  a camera.

adding tee into the video pipeline as described  here

https://github.com/umlaeute/v4l2loopback/issues/83

solves the issue for me.. (I am not sure if this is a good solution, but it works)

@@ -405,6 +405,7 @@
       ! $GST_VIDEO_CONVERTER \
       ! videoscale \
       ! $GST_VIDEO_CAPS \
+      ! tee \
       ! v4l2sink device="$DEVICE" sync=$SYNC
 }

s0170071, 2020-11-21:
tried that, didn't solve the problem.

bluezio, 2021-04-12:
Yes, this is an artifact of the encoding being used, I believe :-). I'll document the need to use multiples of 16 in the `run-videochat.sh` script.

closed 2021-04-12 by bluezio

mail4me9999, 2021-04-12:
I dp/did have video as a multiple of 16 and I need the "tee" for video to work correctly... so there might be something else...the video is  -h 480 -w 864
With the "tee" the video setup is quite reliable.there are problems with the audio the cause of which I cannot pinpoint..wrong device is attached, no audio from the conference when ipwebcam is used.cannot even understand how to recover from this audio problem other than reboot.

414n, 2021-07-17:
I created #112 which adds the `tee` filter to gst pipeline as a workaround with a cli flag (`-t/--with-tee`).