agarciadom/ipwebcam-gst #98: The IP Webcam video feed is not reachable at http://127.0.0.1:2323/video via ADB

m19u31, 2020-10-26:
"The IP Webcam video feed is not reachable at http://127.0.0.1:2323/video.
Please install and open IP Webcam in your phone and start the server.
Make sure that values of variables IP, PORT, CAPTURE_STREAM in this script are equal with settings in IP Webcam."

But all configs is ok! If i open the http://127.0.0.1:2323/video firefox play video. If i open http://127.0.0.1:2323/video in OBS work too. 

ip webcam app dont compatible my device. I use "IP Camera" https://play.google.com/store/apps/details?id=com.shenyaocn.android.WebCam . please help me :)

Y-Shy, 2020-11-04:
I have the same problem on Manjaro Linux.

Video and audio can be received through Firefox, and IP, PORT, CAPTURE_STREAM are setted properly in script, but both ADB and WIFI cannot work (I've tried --use-wifi to specified wifi link), with error "The IP Webcam video feed is not reachable at".

NunoVilaca, 2020-12-04:
Same issue, on ubuntu 18.04.1

bluezio, 2020-12-04:
This script has not been designed for the "IP Camera" app you mention, and that app is not supported. Sorry!

NunoVilaca, 2020-12-04:
> This script has not been designed for the "IP Camera" app you mention, and that app is not supported. Sorry!

I'm using the [IP Webcam](https://play.google.com/store/apps/details?id=com.pas.webcam) app you mention on the main page.

NunoVilaca, 2020-12-04:
Ok, in my case the problem was having an security credentials on the Android App. Is there a way to pass the login and password to the script?

m19u31, 2021-01-18:
> This script has not been designed for the "IP Camera" app you mention, and that app is not supported. Sorry!

ok, but the "ip webcam" app doesn't run on my xiaomi. Do you know if there is another app that is compatible with your script?

bluezio, 2021-04-12:
Just adding this comment before I close, in case somebody else has this issue :-).

@NunoVilaca It may be a matter of tweaking the `BASE_URL` line in the `run-videochat.sh` script, using something like:

```
BASE_URL="http://user:password$IP:$PORT
```

I haven't tried it myself, though.

@m19u31 Sorry, I think this script is only compatible with the mentioned app.

closed 2021-04-12 by bluezio