agarciadom/ipwebcam-gst #99: Not working: module initialisation failed

sm8ps, 2020-11-09:
I had ipwebcam-gst successfully running up to last friday. Come this monday and it is not working anymore. Talk about monday morning blues! I used to have the following script in use for weeks:
```
bash -c "sleep 5 && cvlc --fullscreen --aspect-ratio 16x9 v4l2:///dev/video0" &
run-videochat.sh -w 1920 -h 1080 -f rotate-180
```
Now VLC refuses to open the camera stream stating "v4l2 demux error: not a video capture device" and the same for `stream` instead of `demux`. I followed the trouble shooting section in 'install-videochat.sh' and was confronted with an error that the module v4l2loopback was not available for the current kernel version which was 5.4.0-51. I may have had a kernel upgrade in the meantime but even after a reboot that error persisted.

So I installed v4l2loopback-dkms and could continue the trouble-shooting section. It changed the video device from video0 to video2 which persisted across reboots. The command `gst-launch-1.0 videotestsrc ! v4l2sink device=/dev/video2 & mplayer -tv device=/dev/video2 tv://` then fails with the following error (translated):
```
Connection set to PAUSED ...
ERROR: connection does not want to be paused.
ERROR: be element /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0: »/dev/video2« is not a video playing device.
Additional diagnosis information:
v4l2_calls.c(636): gst_v4l2_open (): /GstPipeline:pipeline0/GstV4l2Sink:v4l2sink0:
Capabilities: 0x85208000
Connection ist set to NULL ...
Connection is emptied ...
```
and then MPlayer fails, too.

When running `bash -x ./run-videochat.sh` I can catch the first error:
```
++ pactl load-module module-echo-cancel sink_name=ipwebcam_echo_cancel source_master=ipwebcam.monitor sink_master=auto_null format=S16LE rate=44100 channels=1 aec_method=webrtc save_aec=true use_volume_sharing=true
Failed: module initialisation failed
```
Zenity shows the standard message:
```
IP Webcam audio is streaming through pulseaudio sink 'ipwebcam'.
IP Webcam video is streaming through v4l2loopback device /dev/video2.
You can now open your videochat app.
```
However, after finishing by hitting Return, the following error is shown:
```
+ pactl unload-module
You must specify an index value for a module or a name
```
Any help, please?

sm8ps, 2020-11-11:
After some extended fiddling I can say that one part of the problem definitely lies at the `pactl` commands. Connection to http://127.0.0.1:8080/videofeed does produce the video stream, however the corresponding audio device does not show any signal when looked at in Pavucontrol. VLC still refuses to open the respective video device, claiming that "not a video capture device".

akuhnregnier, 2020-11-11:
I had a very similar issue this morning (with many identical error messages) after previously having updated some things. I managed to fix it by upgrading v4l2loopback.
See the following links:
- https://askubuntu.com/questions/1289440/v4l2loopback-device-not-working-with-gstreamer?newreg=f78d2266c023453e99be0e9cbfeb929d
- https://github.com/umlaeute/v4l2loopback/issues/370
- In particular, this comment: https://github.com/umlaeute/v4l2loopback/issues/370#issuecomment-723228590, which is a better solution than adding new repositories.

sm8ps, 2020-11-12:
I can confirm with great relief that upgrading to version 0.12.5 does indeed solve the problem. It was easiest for me to upgrade the whole system to Ubuntu 20.10, so there was a whole lot of changes. However, together with the report by @akuhnregnier I think this is confirmed.

Thanks a lot to @akuhnregnier for showing a way out of what seemed a total dead-end to me!

closed 2020-11-12 by sm8ps