agarciadom/ipwebcam-gst #100: [Guide] How to get this working on ubuntu 20.04/linux mint 20

aquaspy, 2020-11-12:
After reading a lot of comments here in issues at github, I finally found an way to make it works at ubuntu 20.04 and mint 20 (which is based on 20.04)

Basically, there are two problems here:

1- The v4l2loopback dkms package that is in ubuntu 20.04 repository (which mint 20 also uses) have a bug that is only fixed in the vesion 0.12.5-1 (which is present at ubuntu 20.10 repositories).

2- For chromium based browsers and cheese, you need to start the module with an parameter.

So, to solve 1, simply download this package: http://za.archive.ubuntu.com/ubuntu/pool/universe/v/v4l2loopback/v4l2loopback-dkms_0.12.5-1_all.deb

and install it on your ubuntu 20.04 or mint 20. You will see a warning saying that it is recommended to install the repository version, but you should ignore it. To install, just go to the folder and use apt install ./v4l2loopback-dkms_0.12.5-1_all.deb or dpkg -i v4l2loopback-dkms_0.12.5-1_all.deb (both with sudo)

Alright, now you can test and see that video is working at firefox.

Now, to make it work in chromium and programs like cheese, unload the module with:
sudo modprobe -r v4l2loopback
and start it again with: sudo modprobe v4l2loopback exclusive_caps=1

now run the command that you run to start ip webcam (./run-videochat.sh --use-wifi 192.168.1.147 --video in my case, just see the ip in your phone. --video makes it use video only, instead of both audio and video)

And now test it :)

Hope it helps.

I may create a PR and organize this guide to make it available at readme, if the owner of this repository approves ;)

bluezio, 2020-11-12:
Of course! Feel free to add a section about Ubuntu 20.04 to the main README if you like.

I'd suggest touching up the module loading options (e.g. maybe adding a file in `/etc/modprobe.d` with `options v4l2loopback exclusive_caps=1`), to make this fix permanent. That way you won't have to remove + reload the module on future boots.

s0170071, 2020-11-21:
Thanks, worked for me. 
Also watch the audio settings. If IPCam is set to not transfer audio, this has also to be disabled in the script. Otherwise it will throw a not found error, but it doesn't mention the cause.

yannisdran, 2021-01-28:
For me in Ubuntu 20.04 it says:
sudo ./install-videochat.sh
sudo: ./install-videochat.sh: command not found
how to fix it?

bluezio, 2021-04-12:
Thanks, I have pushed a change that sets V4L2_OPTS to include `exclusive_caps=1` by default. This fixes the problem when it would not show up in Cheese.

closed 2021-04-12 by bluezio

ElectroGehirn, 2023-08-10:
> For me in Ubuntu 20.04 it says: sudo ./install-videochat.sh sudo: ./install-videochat.sh: command not found how to fix it?

sudo bash ./

and then you should be ready to go