agarciadom/ipwebcam-gst #105: phone not found

manouchk38, 2021-03-16:
I get the following message when running run-videochat:
```
adb is available, but the phone is not plugged in.
Connect your phone to USB and allow usb debugging under developer settings or use Wi-Fi (slower).
```
If I use the wifi option, I get:

```
The IP Webcam audio feed is not reachable at http://192.168.68.100:8080/audio.wav.
The IP Webcam video feed is not reachable at http://192.168.68.100:8080/videofeed.
Please install and open IP Webcam in your phone and start the server.
Make sure that values of variables IP, PORT, CAPTURE_STREAM in this script are equal with settings in IP Webcam.
```

This is atrange because here video stream is find in http://192.168.68.100:8080//video.
Am I  using the same app?

bluezio, 2021-04-12:
Not sure, to be honest: I have just tried the app and it has worked fine for me. It could be a different app. Can you post the link to the Play Store for the app?

mail4me9999, 2021-04-12:
i use ipwebcam via wifi and the app provides video stream on /videofeed and /video at least the versions I use..
I do not know the difference between the two endpoints but both work for me.
I have setup password in ipwebcam and this script happily handles this as well..
(give username:password@ipaddress to the -i switch)

s644, 2021-07-21:
I had the problems via wifi since my latest linux mint upgrade. Requesting just the video stream with `-v` worked for me.

bluezio, 2022-02-24:
Closing due to inactivity.

closed 2022-02-24 by bluezio