agarciadom/ipwebcam-gst #108: Ending the stream breaks subsequents runs of the script

N8n5h, 2021-04-14:
Hi, this script works pretty nice the first time but after ending the stream (by hitting the return key) and trying to run it again (tested with and without any flag) then the webcam "works" but just outputs a frozen frame of the previous stream (presumably the last one) until I reboot to make it work again. 

Tested on Ubuntu 20.04.2 / Firefox + Discord / Chromium + Google Meet

By going to the https://ip:port I can see it's working and updating correctly, so I assume it may have to do with gst-launch ?

vit-zikmund, 2023-02-01:
This could be the cause of the v4l2loopback device still being opened for reading, see the detail I provided at the bottom of https://github.com/agarciadom/ipwebcam-gst/issues/114#issuecomment-1412683287.

agarciadom, 2023-05-03:
This should be handled now, after @vit-zikmund 's PR #123 got merged. (Thanks again for your contribution!)

closed 2023-05-03 by agarciadom