agarciadom/ipwebcam-gst #113: Camera output freezes when Ctrl+C is used instead of <enter> to end the stream

nikora-hakim, 2021-07-26:
I was thinking that I am encountering this https://github.com/bluezio/ipwebcam-gst/issues/108 issue.

After some substantial amount of testing, I found out that it happens only when I hit Ctrl+C instead of <enter>.

**Problem**
When I run script and then kill it with ctr+c, when I run the script next time the video got stucked at last frame, appearing to be frozen.

**Steps to reproduce**
1. Connect phone with USB debuggin on
2. Start script by running `./run-videochat.sh -w 1080`
3. Open some camera tool (for example [`cheese`](https://manpages.ubuntu.com/manpages/bionic/man1/cheese.1.html))
4. Camera should be working
5. Now hit Ctrl+C to kill the scprit.
6. Close `cheese`
7. Start script again
8. Start `cheese` again
9. **the video is now frozen**

The only way how I was able to fix it was to restart my PC.

**Expected behavior**
Script should handle SIGINT signal and terminate gracefully same way as it does when <enter> is used to end the script.

agarciadom, 2023-02-03:
I think this is related to the PR in #121, although I would rather avoid introducing a second copy of the script.

vit-zikmund, 2023-02-15:
Just implemented this with commit https://github.com/agarciadom/ipwebcam-gst/pull/123/commits/b533302ddd93a7a7c104760d26623a754947f810, within PR https://github.com/agarciadom/ipwebcam-gst/pull/123.

agarciadom, 2023-05-03:
I've finally found time to review PR #123, and I've just merged it. Closing it as fixed by that PR - feel free to reopen if that didn't fix your issue.

closed 2023-05-03 by agarciadom