agarciadom/ipwebcam-gst #114: can you make versions for fedora

rebok232yt, 2021-09-10:
i want install this driver and i can't becouse this driver is writed for debian

bluezio, 2021-09-10:
Sorry, I don't use Fedora, and I don't have time for it. Perhaps someone else could pick this one up :-).

vit-zikmund, 2023-02-01:
Hi, I've just made it work under Fedora 37, well, not the script per-se, but the gstreamer pipeline for video, which I'm only interested in. That's the core thing anyway.

Making the run script work would need some more bash sculpting, but... a PR might land maybe later :stuck_out_tongue:
Since I had no previous experience with the technology involved, these were 3 painful days, mostly figuring out why the gstreamer pipeline wont start. But I've cracked it!
 
 TL;DR the (minimal working) gstreamer pipeline has to be:
 ```sh
 gst-launch-1.0 -e -vt --gst-plugin-spew \
  souphttpsrc location="http://$IP:$PORT/videofeed" do-timestamp=true is-live=true \
  ! queue \
  ! multipartdemux ! "image/jpeg,framerate=30/1" \
  ! decodebin \
  ! videoconvert \
  ! v4l2sink device="$DEVICE"
```

All in all the pipeline is pretty much the same as in the script, however, there's an imposed `framerate` cap for the stream coming from the `multipartdemux` element, which needs to be something sensible in the range `[ 1/1, 1000/1 ]`, which is the real `framerate` required by the `v4l2sink` element (targeting the `v4l2loopback` device).
(This can be seen in the log with env var `GST_DEBUG=4`).

Although all the `gst-inspect-1.0` for all the used elements says it pretty much doesn't care about the `framerate` (`framerate: [ 0/1, 2147483647/1 ]`), it's just false information.

To add some flipping and resizing one still can insert these behind `videoconvert`:
```
  ! videoflip video-direction=$FLIP_DIR \
  ! videoscale ! video/x-raw,width=$WIDTH,height=$HEIGHT \
```

FYI: `videoflip`'s `method` option is now deprecated in favor of `video-direction` (see `gst-inspect-1.0 videoflip`).
FYI2: You might notice that the original script forces the `video/x-raw,format=YUY2` at the end, which can still be used, but I haven't noticed it to be required. The `format` negotiated for the `v4l2sink` in my case is `I420`, which works in VLC, OBS Studio and Google Meet.

---

I also think it's worth mentioning how this integrates with the `v4l2loopback` driver, as not understanding that made me waste a day hacking the gstreamer pipeline while the device was used by OBS (and hitting cryptic errors).

The `v4l2loopback` device holds a `format` (`cat /sys/devices/virtual/video4linux/$DEVICE/format`), which defines these items:
- pixel format (that fourcc `YUY2` or whatever)
- width
- height
- fps / framerate

On an unused device, this format is empty (unless you persist it with something like `v4l2loopback-ctl set-caps "video/x-raw, format=YUY2, width=640, height=480" $DEVICE`, which you likely don't want to do). So, when empty, it gets inferred from the stream being written there and is held as long as something is writing and/or something **is reading it**. So when the device is being read by VLC or OBS, despite the gstreamer pipeline is not writing - is dead, the format holds. When you want to restart the pipeline, `v4l2sink` has to negotiate the exact same parameters or you'll end up with
```
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Internal data stream error.
...
streaming stopped, reason not-negotiated (-4)
```
which seems to be an error with the source, but it really means it didn't manage to negotiate all the required caps from the end of the pipeline through its beginning.

So the rule of thumb is to **always close all the device readers before debugging the pipeline**, because at that moment the device's `format` is removed and the pipeline can change things like the resolution or the screen rotation.

FWIW, all I had to get stuff running was to install:
- `v4l-utils` (`v4l-utils-1.22.1-4.fc37.x86_64 @updates`)
- `v4l2loopback` (`v4l2loopback-0.12.7-2.fc37.noarch @rpmfusion-free`)

closed 2023-02-03 by agarciadom