agarciadom/ipwebcam-gst #124: Cannot establish connecttion via WIFI

wmyfelix, 2023-02-27:
` run-videochat --use-wifi 192.168.1.129 -d /dev/video0 -v`

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Got context from element 'souphttpsrc0': gst.soup.session=context, session=(GstSoupSession)NULL;
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink.GstProxyPad:proxypad0: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:src: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstTypeFindElement:typefind.GstPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstGhostPad:sink: caps = image/jpeg
/GstPipeline:pipeline0/GstDecodeBin:decodebin0/GstJpegDec:jpegdec0.GstPad:src: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
/GstPipeline:pipeline0/GstDecodeBin:decodebin0.GstDecodePad:src_0.GstProxyPad:proxypad1: caps = video/x-raw, format=(string)I420, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)jpeg, colorimetry=(string)1:4:0:0, framerate=(fraction)0/1
ERROR: from element /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/libs/gst/base/gstbasesrc.c(3132): gst_base_src_loop (): /GstPipeline:pipeline0/GstSoupHTTPSrc:souphttpsrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: from element /GstPipeline:pipeline0/GstQueue:queue0: Internal data stream error.
Additional debug info:
../gstreamer/subprojects/gstreamer/plugins/elements/gstqueue.c(992): gst_queue_handle_sink_event (): /GstPipeline:pipeline0/GstQueue:queue0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.122299703
Setting pipeline to NULL ...
Freeing pipeline ...
```

vit-zikmund, 2023-03-02:
Hi @wmyfelix, it seems you might have been bitten by the same thing I have a couple weeks ago. [I provided a fix](https://github.com/agarciadom/ipwebcam-gst/pull/122), which has been since merged upstream, but as you are surely using your own fork, you missed it :stuck_out_tongue: 

The `not-negotiated` error means the data types between the pipeline elements could not properly match somewhere on the way from `souphttpsrc` to `v4l2sink`. Sadly, the error messages suck big time and I spent days learning about [debugging] pipelines. In my case (and likely yours), the problem was `framerate=(fraction)0/1`. `souphttpsrc` really only draws a best-effort stream of jpegs, without any notion of a framerate, so at the moment the data becomes a more proper "video", it starts to bear the `framerate=(fraction)0/1` capability. This confuses `v4l2sink`, which [apparently](https://github.com/agarciadom/ipwebcam-gst/issues/114#issuecomment-1412683287) demands something higher than 1. If you force the framerate somewhere in the works, the negotiation succeeds. The fix uses a default of `30/1`, but you can also use an argument to change it as you wish (within the `v4l2sink`'s expectations).

wmyfelix, 2023-03-03:
I tried your fix, and it works!! Also, I have read your description of that problem, and it spotted my situation, especially for the OBS part, which fooled me for a long time. One small flaw is the audio stream doesn't work here on the Pipewire-based PulseAudio, but my old implementation also doesn't work anymore...

closed 2023-03-03 by wmyfelix

vit-zikmund, 2023-03-03:
@wmyfelix Glad I could help ;) Thankfully, I'm not using the audio stream, so I don't need to care so far :relieved: The underlying tools/technologies are obviously far from being mature-stable, so we'll have to live with the change, I guess.