agarciadom/ipwebcam-gst #127: Stream ended unexpectedly!

WaflanCZ, 2024-07-30:
I can connect to my phone but imidietly after that it crashes and says this: Stream ended unexpectedly!

Zeal5, 2024-08-09:
Same warning here!
tried connecting both ways wireless and with cable

adasium, 2024-08-28:
in my case I had a `feed.log`  file  with contents `Error initializing: Unknown option --gst-plugin-spew`
removing this option worked

bermudi, 2024-10-11:
Same here, tried with Xiaomi and Huawei phones and both end unexpectedly right after starting

acora192, 2024-11-17:
So I finally found the issue. I asked chatgpt and the thing is that there was an update over gstreamer and the syntax changed. So you only a line must be changed on the run script:

```bash
JOB_NAME='stream' JOB_LOG=feed.log JOB_INTRO_CMD=stream_intro JOB_FAILURE_CMD=stream_failure \
    monitor_job "$GSTLAUNCH" -e -vt --gst-debug --gst-debug="$GST_DEBUG" $PIPELINE
    # Maybe we need edit this pipeline to transfer it to "Monitor of IP Webcam" to be able to use it as a microphone?
```
Change ```--gst-plugin-spew``` to ```--gst-debug``` on line 643 of run-videochat.sh

agarciadom, 2024-11-18:
Thank you for proposing a fix! I think `--gst-debug` by itself doesn't do much: I just dropped `--gst-plugin-spew` as that is no longer listed in the `gst-launch` manpage:

https://gstreamer.freedesktop.org/documentation/tools/gst-launch.html

commit 0adbffac (2024-11-18): Remove --gst-plugin-spew flag (fixes #127)

The --gst-plugin-spew flag is no longer listed among the options of the gst-launch command:

https://gstreamer.freedesktop.org/documentation/tools/gst-launch.html
closed 2024-11-18 by agarciadom